realsenseai/librealsense #8652: Color stream freezing after 15 frames

aeilkol, 2021-03-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.43.0.3018                    | 
| Language                            |  python                         | 
| Segment			|  other         | 

### Issue Description

I modified the example script to view the camera stream. I put the images into a dictionary indexed by the frame numbers of the frames. After about 15 frames, the color stream freezes, while the depth stream still works. I tested the camera on the opencv_viewer_example.py and it still works without freezing.

This is the initialization of the camera:
```
config = rs.config()

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)

self.pipeline.start(config)
```
This is the loop of the program:

```

align_to = rs.stream.color
align = rs.align(align_to)
while True:
    success, frames = self.pipeline.try_wait_for_frames()
    if not success:
        break
    aligned_frames = align.process(frames)

    if self.background_index == 0:
        self.background_index = frames.frame_number

    depth_frame = aligned_frames.get_depth_frame()
    color_frame = aligned_frames.get_color_frame()

    if depth_frame is None or color_frame is None:
        continue

    depth_image = np.asanyarray(depth_frame.get_data())
    color_image = np.asanyarray(color_frame.get_data())
    number = frames.frame_number

    self.frame_numbers.append(number)
    self.depth_images[number] = depth_image
    self.color_images[number] = color_image

    if len(self.depth_images) > 1:
        print(np.all(np.equal(self.color_images[self.frame_numbers[-1]], self.color_images[self.frame_numbers[-2]]))) <--- this is true after the 15th frame

       self.process_frame()

self.pipeline.stop()
```

MartyG-RealSense, 2021-03-23:
Hi @aeilkol   This problem is known to occur with Python scripts that use the **append** instruction, due to the frames being saved into a **collection**.  The issue is described further in the links below.

https://github.com/IntelRealSense/librealsense/issues/946

https://github.com/IntelRealSense/librealsense/issues/6164

The solution applied by the RealSense user in the first link was to close the pipelines and start them again, whilst in the second link the **Keep()** instruction was used to store the frames in memory.

aeilkol, 2021-03-24:
Thank you! This fixed it for me.

closed 2021-03-24 by aeilkol

MartyG-RealSense, 2021-03-24:
You are very welcome @aeilkol - thanks for the update!