realsenseai/librealsense #8654: can't compile Matlab wrappers on Mac OS X...

drhalftone, 2021-03-23:
Trying to compile 2.43.0 Matlab wrapper and getting the following error at the doorstep of making all:

9 warnings generated.
[100%] Linking CXX shared library librealsense_mex.mexmaci64
clang: error: no such file or directory: '/NODEFAULTLIB:realsense2'
make[2]: *** [wrappers/matlab/librealsense_mex.mexmaci64] Error 1
make[1]: *** [wrappers/matlab/CMakeFiles/librealsense_mex.dir/all] Error 2
make: *** [all] Error 2
dllau@DrHalftone build %

MartyG-RealSense, 2021-03-23:
Hi @drhalftone   The RealSense MATLAB wrapper is only compatible with Windows 10.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/matlab#introduction

drhalftone, 2021-03-23:
Nowhere in the documentation is the word, "only."  I was able to compile the mex file for Mac by commenting out the line that uses the variable '/NODEFAULTLIB:realsense2'.

MartyG-RealSense, 2021-03-23:
Thanks for the information.  The wrapper was designed for Windows 10 though and I am not aware of anyone who has succeeded in building and running it on MacOS.  So there is no advice that I can offer about doing so, unfortunately.

It is also worth bearing in mind that the wrapper currently supports **MATLAB 2017b**, and using newer MATLAB versions such as 2019 and 2020 may result in errors.

drhalftone, 2021-03-24:
Just think, you can now update the wrapper page to reflect this current state of affairs and avoid needlessly wasting your users’ time.

MartyG-RealSense, 2021-03-24:
I sympathize with the time lost.  However, the Windows requirement is stated in the introduction section of the documentation.  

There are ways other than the wrapper to access RealSense data in MATLAB.  If you can provide details of what you are aiming to achieve then I may be able to suggest a suitable method.  Thank you.

closed 2021-03-24 by drhalftone