realsenseai/librealsense #8656: The output point cloud of the API is reversed with the y-axis compared to the Intel Realsense Viewer app (LIDAR CAMERA L515).

anhTuan0712, 2021-03-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.04.01 | 
| Operating System & Version |   Win (8.1/10)   | 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | PC |
| SDK Version                            |   legacy / 2.36.0                           | 
| Language                            |  C/C#/labview/nodejs/opencv/pcl/python/unity                           | 
| Segment			|  { }                          | 

### Issue Description
<Describe your issue 
 1) Here is a picture I took in the app intel Realsense Viewer: AppIntelRealsenseViewer.png
[image]
2) Point cloud is exported from the Intel Realsense Viewer app, I opened it with Meshlap: PointCloudExportFromIntelRealsenseViewer.png 
[image]
3) The point cloud I saved when using the API, I opened it with Meshlap:   PointCloudOutPutOfSDK.png
[image]
4) You can see 2 ply files here:
https://drive.google.com/drive/folders/1P2_mZf-Pfiayd9FH0diYhie5KcQ5_vfZ?usp=sharing
/ question
Why is 3 the opposite of 1 and 2?
 / feature request / etc..>

anhTuan0712, 2021-03-24:
@RealSenseSupport  please help me with this problem! Thank you!

anhTuan0712, 2021-03-24:
@MartyG-RealSense  please help me with this problem! Thank you!

RealSenseSupport, 2021-03-29:
Hi @anhTuan0712 

We will look into it.  Can you outline which API you're specifically using?

anhTuan0712, 2021-03-31:
> Hi @anhTuan0712
> 
> We will look into it. Can you outline which API you're specifically using?

Yes, I follow the following procedure.
step 1: config 
[image]
step2: Get Frame 
[image]
step 3: Save the ply file 
[image]
step 4: Open ply file,  I opened it with Meshlap

anhTuan0712, 2021-04-05:
hi @RealSenseSupport 
I am looking forward to your answer.

RealSenseSupport, 2021-04-10:
Hi @anhTuan0712 

We were unable to reproduce this issue on the L515 and latest development branch.

1.  RealSense Viewer 3D point cloud:
[image]

2. PLY file exported from RSViewer:
[image]

3. PLY file generated using python export ply example here:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py
[image]

4. Also to rule out anything else we instrumented the basic rs-pointcloud C++ example here:
https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud
with the following code to generate a PLY file...
...
auto fname = "_3d_mesh.ply";
std::unique_ptr<rs2::filter> exporter;
exporter = std::unique_ptr<rs2::filter>(new rs2::save_to_ply(fname,pc));
exporter->invoke(depth);
...
...and that matched the other 3 PLY exports above as well (as u can see below):
[image]

Hope this information is helpful in resolving your issue.

Thanks

anhTuan0712, 2021-04-12:
Hi @RealSenseSupport! 
Thank you you have feedback
*** Well when I use ExportToPLY in RS API c # or save_to_ply in RS API C ++ everything is perfect (y-axis pointing up). But when I use the following code to save everything is reversed, I think my code has no problem.
[image]
*** I think in ExportToPLY and save_to_ply, you transformed the y-axis, right?

RealSenseSupport, 2021-04-12:
I'm not sure about your code but why aren't you just using the SDK to generate the PLY file?

anhTuan0712, 2021-04-15:
> I'm not sure about your code but why aren't you just using the SDK to generate the PLY file?

Hi @RealSenseSupport,
Thank you for the feedback.
My job now is to get the point cloud from the SDK and render it to a 3d app but unfortunately everything is reversed, so I tried saving it to flie ply to check and it was reversed as well.
I just want to confirm that at ExportToPLY and save_to_ply, have you made any changes to the y-axis? (like you multiply the y-axis by (-1) ?)

anhTuan0712, 2021-05-10:
Hi @RealSenseSupport , Hi @MartyG-RealSense! 
I am looking forward to an answer from you.

MartyG-RealSense, 2021-07-02:
Hi @anhTuan0712  Sincere apologies that you did not receive a response in this case.  Do you require further assistance with the case, please?  The case will be closed after 7 days from the time of writing this if we do not hear from you.  Thanks!

closed 2021-07-08 by MartyG-RealSense