realsenseai/librealsense #8657: D455 depth data much larger than actual distance when >5m

oxidane-lin, 2021-03-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455                                  | 
| Firmware Version                   | 05.12.09.00  | |
| Operating System & Version  |  Ubuntu 16.04  | 
| Platform                                 | NVIDIA TX2 and PC |
| SDK Version                          | 2.38.1                         |

### issue description

I am now using a D455 as a RGBD camera. However, I found its depth precision worse than I expected. I did a simple test to show some results to you. I placed the camera on the ground, sitting still and placed 2 planes ahead, at a distance of 7.5m. Their relative position is shown below in pic1. 
I recorded about 100 pairs of rgb/depth photos, and did some statistics using my own code. Basically the code reads the depth value of all depth photos at certain pixel point(x,y), record the values and calculate the mean/max/min/standard error. Using my code, I can know the difference between actual distance and returned depth data. 
The results are good when actual distance is less than 4m. But when distances get larger, the depth data get worse. As in my test, the distance is 7.5m. But values of pixels on plane1 are around 8.5-9.5m. I tested with some other objects at the distance of 7.5m, the results are basically the same, which suggested that the values are irrelevant with the plane texture or martial. I added plane2 because I found that the depth data return from plane2 are much closer to 7.5m, which varied from 7.2-8m(still not accurate though). So I have some questions:
1. why does the depth data much larger than real distance?
2. why does the depth data related to the position(or angle) of the plane? Should I do some calibration procedures?

[image]
    pic1: relative position of camera and 2 planes
[image]
    pic2: rgb photo from the camera, showing 2 planes

MartyG-RealSense, 2021-03-24:
Hi @oxidane-lin   The D455 has 2x the depth measuring accuracy over distance of the D435 models, meaning that at 6 meters distance it has the same depth accuracy as the D435 / D435i does at 3 meters.  The distance of 6 meters is the D455's optimal depth accuracy distance, so a reduction in accuracy beyond that distance (7.5 meters) is understandable.  I would not expect the results to deviate as much as the Plane 1 measurements though.

It is possible that the Plane 1 measurements are being affected by the highly reflective floor surface resulting from the sunlight through the windows.  Given that the reflection is weaker where Plane 1 is located though, I would also consider the structure **behind** Plane 1 as a potential cause.

A 400 Series stereo camera can be affected by a phenomenon called **repetitive pattern**, where it can become confused by horizontal or vertical rows of similar objects (for example, a horizontal row of fence posts or tree-tops, or a vertical stack of window blind slats).  The structure behind Plane 1 is similarly repetitive.  

There was a similar scenario with a large room in the case in the link below.  Covering the repetitive structure on the far wall with a cloth helped to break up the pattern and reduce disruption to measurement.

https://github.com/IntelRealSense/librealsense/issues/8258

The more accurate measurements from Plane 2 would seem to validate that the camera is working correctly when there are not potentially disruptive elements included in the camera's field of view (though the railings with regular rectangular sections next to Plane 2 could act as another repetitive element).

The link below discusses the repetitive pattern phenomenon and provides leads to resources for reducing the impact of it.

https://github.com/IntelRealSense/librealsense/issues/6713

oxidane-lin, 2021-03-25:
@MartyG-RealSense  Thank you for your reply but I'm afraid I cannot agree with you about the repetitive pattern thing. As a matter of fact, issue #8258 is also submitted by me. I marked some labels on the picture above to distinguish different areas. Label ABCD are on plane2, label EFG are on " regular rectangular sections next to Plane 2 ", and label HI are behind plane1. As far as I know, the depth values are calculated from disparity, which means the pixel position difference of the same object on left and right camera. The algorithm can mismatch H and I and then calculate wrong depth results. Or mismatch E-F-G, so the depth value on E/F/G are wrong. But it should not mismatch A to E, or C to F, so the depth data on the plane should have nothing to do with the repetitive pattern behind. You can even say there are mismatches among A to B or C or D AND I think that's kind of reasonable.

[image]

I was wandering why you always focus on a distance of 6m, until I find that on your official website that the ideal range is "0.6-6m". I remembered that when I bought the D455 camera, the main factor I cared about is the max valid depth distance can be " up to 20m". But when I recheck this parameter on the official website, nothing like that left. The parameters in the whitepaper of D400 SERIES also use ambiguous descriptions like "Range 0.4m to over 10m (Varies with lighting conditions)", which is almost the same as D435i. So if the two camera D435i and D455 are so much alike, why should I buy both of them? Is D455 only a baseline-larger D435 with global shutters? Did your official parameters **do changed** or I remembered it wrong?

[image]

[image]

MartyG-RealSense, 2021-03-25:
The D455's range is typically described in official Intel information in terms of the **6 meter optimal depth range** or as **0.4 m to over 10 m**, though a number of non-Intel information sources have previously published a 20 m value for the maximum range.

The one official source I know of that does quote the 20 m value for D455 is this one:

https://www.intelrealsense.com/stereo-depth/ 

The full list of advantages of the D455 model is described on the official announcement press release for D455 on the **Intel Newsroom** site.

https://newsroom.intel.com/news/intel-realsense-depth-camera-d455-twice-range-better-performance/#gs.x3yzip

MartyG-RealSense, 2021-03-25:
In regard to stereo matching, you could experiment with the **Second Peak Threshold** setting.  In the RealSense Viewer, this control can be found in the **Advanced Controls** category, in the **Depth Control** sub-section.  This parameter will disqualify a depth pixel if the second-best match is too close to the best match.

[image]

oxidane-lin, 2021-03-29:
@MartyG-RealSense I've tried the parameter 'second peak thr' and cannot it cannot fix the problem...Actually it only decrease the effective depth area when I increase the parameter's value. According to my understanding of 'second peak thr', it only filters out the points where second match is close to the most match, but it cannot change the value of the most match. So even if I increase the parameter, the depth value still stay wrong. Am I right?

MartyG-RealSense, 2021-03-29:
The majority of Advanced Mode functions are not documented, because they interact with each other in complex ways.  So Intel chose to control them with machine learning algorithms, whilst allowing RealSense users to perform experimentation with the values.  There is therefore not enough reference material for me to answer your question about the behaviour of the value.

Intel's **Depth Map Improvements for Stereo-based Depth Cameras on Drones** white paper does provide a definition for the threshold's Second Peak Delta though.

***************

When analyzing the disparities of an area in the stereo images for a match, there might be one clear candidate indicating a large peak in terms of correlation.  In some cases, multiple candidates could be viable at different peak levels.  The second peak threshold determines how big the difference from another peak needs to be, in order to have conﬁdence in the current peak being the correct one. 

****************

https://dev.intelrealsense.com/docs/depth-map-improvements-for-stereo-based-depth-cameras-on-drones#section-b-depth-camera-settings

MartyG-RealSense, 2021-04-06:
Hi @oxidane-lin  Do you have an update about this case that you can provide, please?  Thanks!

oxidane-lin, 2021-04-06:
@MartyG-RealSense Not yet. According to your former reply, I can better understand the concept "second peak thr or delta" but it does little help to my initial problem. The conclusion I can draw is that the distance outside of 6m is not what the camera designed for. Single or several parameters' change will not help so much because the parameters usually interact with each other. 
And I know now you're not a technical engineer from intel. So thank you for your kind help, and you can close this issue for now as I don't have updates. I will reopen it if needed.

MartyG-RealSense, 2021-04-06:
I am a Support Engineer for RealSense at Intel.  

I will close the issue as you suggest.  Please feel free to re-open it at a future date or create a new case if you need to.  Thank you very much.

closed 2021-04-06 by MartyG-RealSense