realsenseai/librealsense #8660: How to combine data from t265 and 435i for localization in the same coordinates

vnadein, 2021-03-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 and 435i } | 
| Firmware Version                   | (2.41.0) | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson /  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am trying to make it so that in the event of an error and a reboot of the system, the system was found in the same coordinates as it was, and the vector axes were not zeroed. I read a lot of things, but I didn't understand how to do it at all, maybe someone solved a similar problem and could share the lines of code for this solution?

RealSenseSupport, 2021-03-26:
@vnadein Could you please check if https://docs.opencv.org/3.4/d4/d94/tutorial_camera_calibration.html helps?

vnadein, 2021-03-26:
> @vnadein Could you please check if https://docs.opencv.org/3.4/d4/d94/tutorial_camera_calibration.html helps?

I thought maybe your library has a solution to this problem, like competitors, this action is carried out using a simple function, if a previously saved card is loaded into the camera.
Since relocalization is present, but how it works and what makes it completely unclear, the pyrealsense library is very poorly described

RealSenseSupport, 2021-03-31:
@vnadein  Sorry we don't have such example based on python. We have one example based on ROS https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping and please check if it helps.  Thanks!

RealSenseSupport, 2021-04-06:
@vnadein Any other questions about this ticket? Looking forward to your update. Thanks!

RealSenseSupport, 2021-04-20:
@vnadein Sorry we will close the issue at this point as we don't hear from you for weeks. Please feel free to create another new ticket if you still have questions or issues. Thank!

closed 2021-04-20 by RealSenseSupport