realsenseai/librealsense #8661: realsense-viewer can not open

987428377, 2021-03-24:
Camera Model :D435i
Method: Building librealsense - Using vcpkg
problem:When I open  realsense-viewer,  one error appeared：
GLFW Driver Error: GLX: Failed to create context: BadValue (integer parameter out of range for operation)
Could not initialize offscreen context!
who can help me？ thanks

MartyG-RealSense, 2021-03-24:
Hello, this error is typically unrelated to librealsense and can appear on a range of software products that makes use of OpenGL graphics.

As a first step in investigating your problem, I would recommend updating the graphics drivers on your computer.

987428377, 2021-03-24:
Thanks! When I restart the computer, it's ready to use.

987428377, 2021-03-24:
> Hello, the error **GLFW Driver Error: GLX: Failed to create context: BadValue (integer parameter out of range for operation)** is likely to be unrelated to librealsense, as it can appear on a range of software products that makes use of OpenGL graphics.
> 
> As a first step in investigating your problem, I would recommend updating the graphics drivers on your computer to see whether it corrects the problem (by installing OpenGL if it is absent or updating it if it is installed but out of date).

Thanks! When I restart the computer, it's ready to use.

MartyG-RealSense, 2021-03-24:
Great to hear - thanks very much for the update!  :)

MartyG-RealSense, 2021-03-30:
Case closed due to solution achieved.

closed 2021-03-30 by MartyG-RealSense