realsenseai/librealsense #8663: No laser emitter on the D455...

drhalftone, 2021-03-24:
I haven't been able to get the laser to turn on in the D455 on either Mac or Windows using the realsense viewer.  It used to work, but stopped working before and after I updated the firmware.

MartyG-RealSense, 2021-03-24:
Hi @drhalftone   Could you check please whether the menu beside the **Emitter Enabled** option in the **Stereo Module** section of the Viewer's options side-panel is set to **Laser** please, and not **Off** or **Laser Auto**.  If this setting was set to Off or Laser Auto during a previous session of the Viewer then the setting may persist in subsequent sessions of the program even after it has been closed and re-launched, until the Laser option is selected again.

[image]

drhalftone, 2021-03-24:
It’s on laser and I’ve manipulated the laser power at various points, usually maxed out at 350.  I always assumed the realsense would turn itself off if the temperature gets to high.

MartyG-RealSense, 2021-03-24:
The laser safety mechanism in the firmware will halve the laser power if the Projector temperature (viewable under the **Controls** section of the Stereo Module options - is detected to be 60 degrees C or more for several seconds, and then shut the laser off if the temperature continues to be at this excessive level after that.

drhalftone, 2021-03-24:
That’s a relief. I’ll dig up the error log and attach it shortly. Maybe you can see something useful.

MartyG-RealSense, 2021-04-06:
Hi @drhalftone  Could you update me about whether you require further assistance with this case, please?  Thanks!

drhalftone, 2021-04-06:
Close it out.  I’m getting the error intermittently inside the viewer that basically says something along the lines of not responsive, but is has an HWERROR which jumps out to me as HardWare Error.  But I don’t seem to be getting it in my own software.

MartyG-RealSense, 2021-04-06:
Do you mean to close the case please?

closed 2021-04-06 by drhalftone