realsenseai/librealsense #8666: rs-record on RPI4 32bit throws rosbag::BagIOException after hitting 2 Gb file limit (=30 s recording)

aviogit, 2021-03-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | I don't think firmware it's the issue here | 
| Operating System & Version |   Ubuntu 20.04 32bit  | 
| Kernel Version (Linux Only)    |  5.10.9-v7l+                                         | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  librealsense commit: 8f355d5e8dfc9096494214127ceecaf125bcbfc7 (22 Jan)                          | 
| Language                            |  C++                           | 

### Issue Description

`rs-record` throws after hitting the 2 Gb file limit (about 30 s worth of data) on a Raspberry Pi 4 with Ubuntu 20.04 32 bit. This doesn't happen with normal `rosbag record` (it has been fixed in 2015: https://github.com/ros/ros_comm/issues/464#issuecomment-70968557).

#> ./rs-record -t 120
Recording t = 119s
Finished
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error writing to file: writing 4 bytes, wrote 1 bytes
Aborted (core dumped)

MartyG-RealSense, 2021-03-24:
Hi @aviogit    Could you clarify please what program you are using when you mention **rs-record** - is it the librealsense example progran **rs-record-playback** or some other tool?

https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp

aviogit, 2021-03-24:
HI @MartyG-RealSense, thank you for the super fast reply. Actually I was trying to record with the [rs-record tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/recorder) because I was having some issue recording a normal bag with `rosbag record` using [rs_d435_camera_with_model.launch](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_d435_camera_with_model.launch) as interface with the camera. I was observing lower framerates (e.g. 20 or so vs. 30 fps nominal) or even worse trying to record `CompressedImage` or `CompressedDepth` topics. My long-term objective would be to use the OMX h264 hardware encoder of the PI4, but I'm realizing it won't be so straightforward as I imagined.

MartyG-RealSense, 2021-03-24:
I have not seen the **rs-record** tool mentioned in some years, hence my non-recognition of the program name.  I do apologize.   Given that rs-record's script has not been updated in over 2 years at the time of writing this, I would recommend instead recording a bag file in the **RealSense Viewer** program using the 'Record' button at the top of its options side-panel.

aviogit, 2021-03-24:
Ok, I'll give it a try, but I still need a command line tool able to capture reliably, I have to setup a system able to capture with a single command or starting at boot or via a web interface (ideally all these options together).

MartyG-RealSense, 2021-03-24:
A popular solution with ROS users for launching a ROS node automatically upon computer boot is **robot_upstart**

https://github.com/clearpathrobotics/robot_upstart

I would also recommend using the roslaunch instruction below instead:

**roslaunch realsense2_camera rs_camera.launch**

In regard to the website control aspect, I wonder if using **port forwarding** to access ROS remotely would work for your project.

http://wiki.ros.org/ROS/Tutorials/MultipleRemoteMachines#PortForwarding_.28PF.29

aviogit, 2021-03-30:
Hi @MartyG-RealSense,

thank you for all your support. The 2 Gb truncation problem is present also in `rs-record-playback`, so the problem should be in the implementation of `src/media/ros/ros_writer.cpp`. Any idea on where I could start looking for this bug?

MartyG-RealSense, 2021-03-30:
You could next try increasing the size of the **swapfile** on your Pi so that it has more virtual memory (space on a storage medium such as SD card) to use once the real memory of the Pi is used up.  For example, define a swapfile size of **4096** instead of 2048.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md#add-swap

aviogit, 2021-03-30:
I'm using a RPI4 with 8 Gb of RAM and 9 Gb of swapspace (0 bytes currently used because `rs-record-playback` is not memory intensive at all). I think the problem is in `src/media/ros/ros_writer.cpp`, there is something that breaks the recording on 32 bit systems after writing 2 Gb of bag file. And if I click `stop recording`, the process crashes with the usual `rosbag::BagIOException`:

> terminate called after throwing an instance of 'rosbag::BagIOException'
>   what():  Error writing to file: writing 4 bytes, wrote 1 bytes
> Aborted (core dumped)

So the rosbags recorded this way are also invalid and need to be reindexed.

MartyG-RealSense, 2021-03-30:
Thanks very much for the additional information.  I will refer your problem to Intel for advice.

MartyG-RealSense, 2021-03-31:
I received advice from Intel about your case that the problem is believed to be related to use of a 32-bit Ubuntu image.  The problem was able to be replicated with 32-bit Raspbian.

The advice added that a 64-bit image is preferred for Pi 4 and pointed to the **32 bit vs 64 bit** section of the page linked to below, under the heading **Which image should I pick?**

https://ubuntu.com/download/raspberry-pi

aviogit, 2021-03-31:
@MartyG-RealSense thanks for your support, but I don't think I'll switch distribution because 32 bit are not supported, I'll switch camera model. I'm having so many problems (like [this one](https://github.com/IntelRealSense/librealsense/issues/7060)) with Realsense and USB on the Raspberry Pi, that I think it's easier to switch camera than trying to find one configuration (out of a thousand theoretically possible) where Realsense works.

closed 2021-04-07 by MartyG-RealSense