realsenseai/librealsense #8668: Limited Resolution

wangziqi000, 2021-03-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (1.5.5.0 --> 1.5.2.0) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/ NVIDIA Jetson |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
Hi,

I experienced the same problem as @LarsBuck, which is described in https://github.com/IntelRealSense/librealsense/issues/8319#issue-802182156

After updating the firmware to the new version (1.5.5.0) recommended in realsense-viewer, the device is then recognized as a USB2.0 device and the only available resolution is 320x240 for depth and 640x480 for RGB. Very weird. I revert the firmware to 1.5.2.0 and it worked as before. But still, I would like to benefit from new firmware updates.

You might want to check if there are problems with the new firmware. @RealSenseSupport
Thanks for any help!

RealSenseSupport, 2021-03-24:
Hi @wangziqi000 

Couple of questions regarding what you're seeing:
- Between FW updates are you unplugging the USBC connector at any point?  
- Is there anyway you can provide rs-enumerate-devices printout for both FWs on your camera?
- Can you provide RealSense Viewer camera info image for your L515 camera when having both FWs on your system?

wangziqi000, 2021-03-31:
Between FW updates are you unplugging the USBC connector at any point?
No I did not.

I will work on the rest two and get back to you by the end of this week.

RealSenseSupport, 2021-04-08:
Sounds good.  Thanks

closed 2021-04-08 by RealSenseSupport

wangziqi000, 2021-05-21:
I happened to be able to reproduce this problem today.
This time switch drivers is no longer helpful. Could the device be faulty?

This is the answer to the questions regarding what I'm seeing:

(1) Between FW updates are you unplugging the USBC connector at any point?
No.

(2) Is there anyway you can provide rs-enumerate-devices printout for both FWs on your camera?

$ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	f0352228
    Firmware Version              : 	01.05.05.00
    Recommended Firmware Version  : 	01.05.05.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:07.1/0000:0b:00.3/usb5/5-3/5-3:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	2.1
    Product Line                  : 	L500
    Asic Serial Number            : 	0003b53bd42c
    Firmware Update Id            : 	0003b53bd42c

Stream Profiles supported by L500 Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Confidence	  320x240	@ 30Hz	   RAW8
    Infrared	  320x240	@ 30Hz	   Y8
    Depth	  320x240	@ 30Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  640x480	@ 30Hz	   RGB8
    Color	  640x480	@ 30Hz	   Y16
    Color	  640x480	@ 30Hz	   BGRA8
    Color	  640x480	@ 30Hz	   RGBA8
    Color	  640x480	@ 30Hz	   BGR8
    Color	  640x480	@ 30Hz	   YUYV
    Color	  640x480	@ 15Hz	   RGB8
    Color	  640x480	@ 15Hz	   Y16
    Color	  640x480	@ 15Hz	   BGRA8
    Color	  640x480	@ 15Hz	   RGBA8
    Color	  640x480	@ 15Hz	   BGR8
    Color	  640x480	@ 15Hz	   YUYV
    Color	  640x480	@ 6Hz	   RGB8
    Color	  640x480	@ 6Hz	   Y16
    Color	  640x480	@ 6Hz	   BGRA8
    Color	  640x480	@ 6Hz	   RGBA8
    Color	  640x480	@ 6Hz	   BGR8
    Color	  640x480	@ 6Hz	   YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 400Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 100Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 100Hz	   MOTION_XYZ32F
    

$ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	f0352228
    Firmware Version              : 	01.05.02.00
    Recommended Firmware Version  : 	01.05.05.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:07.1/0000:0b:00.3/usb5/5-3/5-3:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	2.1
    Product Line                  : 	L500
    Asic Serial Number            : 	0003b53bd42c
    Firmware Update Id            : 	0003b53bd42c

Stream Profiles supported by L500 Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Confidence	  320x240	@ 30Hz	   RAW8
    Infrared	  320x240	@ 30Hz	   Y8
    Depth	  320x240	@ 30Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  640x480	@ 30Hz	   RGB8
    Color	  640x480	@ 30Hz	   Y16
    Color	  640x480	@ 30Hz	   BGRA8
    Color	  640x480	@ 30Hz	   RGBA8
    Color	  640x480	@ 30Hz	   BGR8
    Color	  640x480	@ 30Hz	   YUYV
    Color	  640x480	@ 15Hz	   RGB8
    Color	  640x480	@ 15Hz	   Y16
    Color	  640x480	@ 15Hz	   BGRA8
    Color	  640x480	@ 15Hz	   RGBA8
    Color	  640x480	@ 15Hz	   BGR8
    Color	  640x480	@ 15Hz	   YUYV
    Color	  640x480	@ 6Hz	   RGB8
    Color	  640x480	@ 6Hz	   Y16
    Color	  640x480	@ 6Hz	   BGRA8
    Color	  640x480	@ 6Hz	   RGBA8
    Color	  640x480	@ 6Hz	   BGR8
    Color	  640x480	@ 6Hz	   YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 400Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 100Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 100Hz	   MOTION_XYZ32F

(3) Can you provide RealSense Viewer camera info image for your L515 camera when having both FWs on your system?
See attached.
[image]