realsenseai/librealsense #8675: rs-server Product ID not supported

MichaelStrohmeier, 2021-03-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455, L515 | 
| Firmware Version                   |05.12.12.100 / 01.05.05.00| 
| Operating System & Version | 20.04 (PC) / 18.04 (Jetson) | 
| Kernel Version (Linux Only)    | 5.8.0-45-generi / 4.9.140-tegra | 
| Platform                                 | PC / NVIDIA Jetson Xavier NX |
| SDK Version                            |  v2.43.0                  | 

### Issue Description
I try to stream RGB and depth data from a Jetson Xavier NX to my PC.
I compiled the librealsense on both platforms from source, with respective flags:

PC:
`cmake .. -DBUILD_NETWORK_DEVICE=ON -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=ON`

Jetson:
`cmake .. -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true -DBUILD_NETWORK_DEVICE=ON`

If any depth sensor is connected to PC/Jetson directly, **realsense-viewer** is working as expected.
On either platform **rs-server** starts without errors/warning.
If I try to connect using **realsense-viewer** as client, I get the following error Message:
`info Product Id not supported by the device!`

I narrowed it down to **common/model-view.cpp** line 3820:
`if (s->is<color_sensor>() || !strcmp(dev.get_info(RS2_CAMERA_INFO_PRODUCT_ID), "0B5B"))
                objects = _detected_objects;`

3dsf, 2021-03-28:
I'm experiencing a similar issue.  

> Invalid Value in rs2_get_device_info(dev:0x3521800, info:Product Id):
> info Product Id not supported by the device!

I see the camera is connected via a notification in the top right corner.

After that I, can see the device details in the **Add Source** drop down and I'm getting a repeated info 

> backend-v4l2.cpp:502 - Cannot access /sys/class/video4linux

---   

- jetson nano
   - default nvidia image
      - librealsense install from Oct 2020 using special install UVC build (rs-server confirmed working at the time)
- pc install
   - Fedora 34 
      - librealsense install w/ deleted line mentioned in #6897 to allow librealsense with rs-server to compile

3dsf, 2021-03-28:
When compiled with _-DFORCE_RSUSB_BACKEND=ON_ on the PC, it still does not work, but the errors in a different way 

Still get and Invalid Value error
> Invalid Value in rs2_get_device_info(dev:0x41e16e0, info:Product Id):
> info Product Id not supported by the device!

and realsense-viewer sees the camera  

**Info**
> context.cpp:382 - Found 1 RealSense devices (mask 0xff)
> notifications.cpp:514 - Intel RealSense D435I IP Device Connected
> ip_device.cpp:258 - Option 'Error Polling Enabled' has changed to: 1

**Warning**
> output-model.cpp:870 - Unrecognized command ''

RealSenseSupport, 2021-05-06:
@3dsf Did you have a chance to try v2.43-44?

3dsf, 2021-05-12:
@RealSenseSupport 
builds with rs-sever enabled, thx

closed 2021-05-25 by RealSenseSupport

AndreV84, 2021-05-27:
@RealSenseSupport Does it support or doesnt the l515?
https://github.com/IntelRealSense/librealsense/issues/9014#issuecomment-841533559