realsenseai/librealsense #8676: Save depth-frame to ros-bag, with color-frame accessible only real-time

indridieinarsson, 2021-03-26:
--------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.11.00 | 
| Operating System & Version |   raspberry OS | 
| Kernel Version (Linux Only)    | 5.4.79                                        | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  2                          | 
| Language                            |  C++ / Python                          | 
| Segment			|  other                         | 

### Issue Description
I am working with a Realsense D455 camera. I need to save the depth-frame stream to a file, as it is needed for further processing. This limits my choice to no compression or lossless compression. On the other hand, while the RGB stream is also needed, it is only for visual evaluation, no processing will be done on color frames. Therefore, conventional video compression is fine for the RGB stream.

It is well established, that when saving to a ros-bag, the file size get quite huge very fast. I'm trying to figure out some ways to reduce the file sizes. Optimally, I would record with my minimal requirements : depth losslessly, rgb lossy.

Now, librealsense offers only lossless compression (and only in C++), which doesn't save a lot of space.

What I want to do, is to have librealsense record only the depth-frames to a ros-bag file (possibly compressed), and handling the color-frame myself via opencv. But when I enable recording to ros-bag, librealsense records both streams to the bag file.
Is there a way to have librealsense do this, enable both streams, but divert only the depth-frame to the bag file while handing the other to me in code? I am comfortable with solutions in C++ or python (though preferring python). 

Another solution would be to handle everything myself - separating the depth- and color-frames, and saving the depth frames to a binary format myself.  I tried doing this with HDF5, but had throughput problems - I guess the rosbag code path in librealsense is well tuned for this use case (IO throughput is good enough to save to rosbag, so should not be the limiting factor). If someone has a hint for something that might work, I would be thankful.

MartyG-RealSense, 2021-03-26:
Hi @indridieinarsson   If you are able to record a bag with **ROS** using its **rosbag record** function then that could provide a means to make file size manageable.  This is because rosbag record has a function called **split** that automatically stops the recording of the current file and starts a new file when a defined condition is reached, such as a maximum file size or a recording duration.

http://wiki.ros.org/rosbag/Commandline#rosbag_record

[image]

Alternatively, some RealSense users access a RealSense camera's RGB using a video streaming tool called **GStreamer**.  The link below has an example of a GStreamer plugin for RealSense that has been tested with the RealSense D435i model.

https://github.com/WKDSMRT/realsense-gstreamer

A RealSense Python user who also wanted to avoid huge bag files took the approach of recording both depth and RGB as **.avi** video files, though this approach has the complication of reading depth data from the depth avi once the recording has been made.

https://github.com/IntelRealSense/librealsense/issues/8528

Someone who also wanted to access avi data in librealsense was offered the suggestion that they could read the avi frames in OpenCV and then send them to librealsense's **software_device** synthetic frames interface.

https://github.com/IntelRealSense/librealsense/issues/2249#issuecomment-413492867

**software_device**
https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

MartyG-RealSense, 2021-04-12:
Case closed due to no further comments received,

closed 2021-04-12 by MartyG-RealSense