realsenseai/librealsense #8677: SR305 post processes the depth image

LadaOndris, 2021-03-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |   Arch Linux  | 
| Kernel Version (Linux Only)    |  5.11.8                                   | 
| Platform                                 | PC  |
| SDK Version                            |   2.36.0.2038                          | 
| Language                            |  python                         | 
| Segment			|  Hand Detection, Hand Pose Estimation     | 

### Issue Description

Hey, I am using the SR305 depth camera to retrieve depth image. I perform hand detection and pose estimation from these images. However, the camera has started giving me postprocessed images. Some time ago, it worked just fine. I don't know what has changed or if it is configurable, but I didn't find anything in the documentation. 

The following image demonstrates what the camera is giving me. It basically removes everything except my head and hand. 

[image]

I see it performs some kind of hand segmentation. Can I disable this behavior? I use the pyrealsense library in Python.

Thanks a lot.

MartyG-RealSense, 2021-03-26:
Hi @LadaOndris   A self-created script should not be applying any post-processing filters, as they have to be deliberately programmed into the application (unlike the **RealSense Viewer** program, which applies a range of presets and post-processing filters by default upon launch).

Does your script set any of the SR300 / SR305 **Visual Presets** such as 'Background Segmentation'?

LadaOndris, 2021-03-26:
Thanks for the response. I do not set any Visual Presets. 
I use the following code:
```
def generate_live_images():
    pipe = rs.pipeline()
    cfg = rs.config()
    cfg.enable_stream(rs.stream.depth, 640, 480)
    profile = pipe.start(cfg)
    try:
        while True:
            frameset = pipe.wait_for_frames()
            depth_frame = frameset.get_depth_frame()
            depth_image = np.array(depth_frame.get_data())
            depth_image = depth_image[..., np.newaxis]

            yield depth_image

    finally:
        pipe.stop()
```
Are there any configuration settings I might check?

MartyG-RealSense, 2021-03-26:
It does not look as though there is much that can go wrong with the script as it is short and simple.  It may be worth checking the camera with another Python script though.  A good test script would be the SDK's **opencv_viewer_example** script.  This script also stores the data in a numpy array, but takes the approach of using **np.asanyarray** instead of **np.array**.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py#L43

LadaOndris, 2021-03-26:
@MartyG-RealSense I ran the example you said, and it worked just fine. Then I tried my script and now it worked as expected! I don't quite understand what's reason behind it. Anyway, I'm grateful that it's working. By the way, both function **np.array** and **np.asanyarray** work. If the problem occurs again in the next few days, I'll let you know. Thanks!

MartyG-RealSense, 2021-03-26:
Great to hear @LadaOndris that your program is working well again!  Yes, please do update here in the coming week about whether the problem re-occurs or whether it remains working correctly.  Good luck!

MartyG-RealSense, 2021-04-06:
Hi @LadaOndris  Do you require further assistance with this case, please?  Thanks!

LadaOndris, 2021-04-06:
Hi @MartyG-RealSense, the problem did not reoccur. It works correctly.

MartyG-RealSense, 2021-04-06:
Great to hear, @LadaOndris - thanks for the update!   I will close this case now as you have had a successful outcome.  Thanks again!

closed 2021-04-06 by MartyG-RealSense