realsenseai/librealsense #8678: Rs-Converter filename timestamps dont match ros message timestamps

geoeo, 2021-03-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | 2.40 | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
This is a issue with rs-converter in tools, not with the specific realsense api.
The timestamps in the image conversion are a factor 1000 greater than the ros message timestamps, this makes it confusing for user to write their own conversion, especially since imu conversion is not done via the rs-converter tool. 

[image]
[image]

Either the timestamps of the filenames match the ros message timestamps or the user is informed that this is the case in the Readme

MartyG-RealSense, 2021-03-27:
Hi @geoeo   I have applied the **Documentation** tag to this case so that it can be kept open and your suggested change to the readme of **rs-convert** considered by the RealSense team at a future date.

In regard to support for IMU data in rs-convert, an official feature request has already been filed to consider adding this feature to the tool.

https://github.com/IntelRealSense/librealsense/issues/8320

geoeo, 2021-03-27:
Thanks a lot. On another note, the rs-converter tool seem to be discarding the first image in the rosbag, as can be seen from the screenshots. Should I open another issue for this?

closed 2026-10-23 by MartyG-RealSense