realsenseai/librealsense #8679: Python: rs.colorizer and depth_colormap

czy10383, 2021-03-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L515  | 
| Firmware Version                   | 01.05.04.01 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  NA                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2                        | 
| Language                            |  python                           | 
| Segment			|   others                         | 

### Issue Description

I have generated the depth_colormaps using rs.colorizer

depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())

I am also using a python 3 queue (Q) to save the depth_colormaps

Q.put(depth_colormap)

When I run my program, it gives me a RuntimeError

depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
RuntimeError: Error occured during execution of the processing block! See the log for more info

The RuntimeError does not occur if I do either of the following:

1. if I change the code to
depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data()).copy()

or

2. If I switch to using opencv instead

depth_colormap2 = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=alpha), cv2.COLORMAP_JET)

May I ask what is the issue that caused the RuntimeError? My code is below. Thank you

```
import cv2
import pyrealsense2 as rs
import numpy as np
from queue import Queue

colorizer = rs.colorizer(color_scheme=0)
try:
    # create a queue
    Q = Queue(maxsize=5*30)

    # Create a context object. This object owns the handles to all connected realsense devices
    pipeline = rs.pipeline()

    # Configure streams
    config = rs.config()
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

    # Start streaming
    pipeline.start(config)

    alpha = 0.03

    while True:
        # This call waits until a new coherent set of frames is available on a device
        # Calls to get_frame_data(...) and get_frame_timestamp(...) on a device will return stable values until wait_for_frames(...) is called
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        if not depth_frame:
            continue

        # depth_colormap0 is a [480 x 640] ndarray
        depth_image = np.asanyarray(depth_frame.get_data())

        # depth_colormap is a [480 x 640 x 3] ndarray
        depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
        # depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data()).copy()

        depth_colormap2 = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=alpha), cv2.COLORMAP_JET)

        if Q.full():
            dropped_frame = Q.get()
            print('queue full...dropping earliest frame')
        Q.put(depth_colormap)

        images = np.hstack((depth_colormap, depth_colormap2))

        cv2.imshow(f'output', images)

        key = cv2.waitKey(1) & 0xFF
        if key == ord("q"):
            cv2.destroyAllWindows()
            break
    exit(0)

except Exception as e:
    raise e
    pass
```

RealSenseSupport, 2021-05-19:
This is an issue about object lifetime.

The function np.asanyarray will a reference to the memory in the frame object.
The other method will copy the memory to another array.

https://stackoverflow.com/questions/14415741/what-is-the-difference-between-numpys-array-and-asarray-functions
https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_lifetime.md

MartyG-RealSense, 2026-10-23:
This open GitHub issue relating to the RealSense L515 camera is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense