realsenseai/librealsense #8680: Passing Java Frame to JNI function asynchronously

idenc, 2021-03-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  NA                              | 
| Platform                                 | Android  |
| SDK Version                            |  {2,43                           | 
| Language                            | Java/C++}                          | 
| Segment			|  Smartphone                        | 

### Issue Description
I have a pipeline loop that is running on Android and displaying frames on the screen. Now I want to do some processing on the pointcloud asynchronously in C++ since I am using the PCL library to do some pointcloud operations. However, no matter what I try I encounter a Fatal signal 11 (SIGSEGV) as below 
`A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 2976 (Thread-11), pid 1616 (com.idenc.toph)`

The code works if run synchronously but this blocks the display from being smooth. I have a hunch that it is the frame being released so the memory is invalid when the C++ code tries to access it. Is there a way to deep copy the frame? I'm not sure what clone does because it seems to still be pointing to the same memory. Here is the relevant code:

### Android
```
try (FrameReleaser releaser = new FrameReleaser()) {
    FrameSet frames = mPipeline.waitForFrames().releaseWith(releaser);
    FrameSet aligned = frames.applyFilter(align).releaseWith(releaser);
    FrameSet decimated = aligned.applyFilter(decimationFilter).releaseWith(releaser);
    FrameSet holeFilled = decimated.applyFilter(holeFillingFilter).releaseWith(releaser);
    FrameSet processed = holeFilled.applyFilter(mColorizer).releaseWith(releaser);

    VideoFrame colour = processed
            .first(StreamType.COLOR)
            .releaseWith(releaser)
            .as(Extension.VIDEO_FRAME);
    mGLSurfaceView1.upload(colour);

    Frame depth = processed
            .first(StreamType.DEPTH)
            .releaseWith(releaser);
    mGLSurfaceView2.upload(depth);
    try (DepthFrame depthFrame = holeFilled.first(StreamType.DEPTH).releaseWith(releaser).as(Extension.DEPTH_FRAME)) {
        if (!pclInitialized) {
            DepthInterface.initializePCL(depthFrame, colour);
            pclInitialized = true;
        }
        if (!isRunning) {
            isRunning = true;
            new Thread(() -> {
                DepthInterface.performGroundSegment(depthFrame.clone().as(Extension.DEPTH_FRAME), colour.clone().as(Extension.VIDEO_FRAME));
                isRunning = false;
            }).start();
        }
    }
}
mHandler.post(mStreaming);
} catch (Exception e) {
Log.e(TAG, "streaming, error: " + e.getMessage());
e.printStackTrace();
}
```

C++ Code

```
extern "C" JNIEXPORT void JNICALL
Java_com_idenc_toph_DepthInterface_nPerformGroundSegment(JNIEnv *env,
                                                         jclass type,
                                                         jlong depth_handle,
                                                         jlong color_handle) {
    auto depth_frame = rs2::frame(reinterpret_cast<rs2_frame *>(depth_handle));
    auto color_frame = rs2::frame(reinterpret_cast<rs2_frame *>(color_handle));

    rs2::pointcloud pointcloud;

    pointcloud.map_to(color_frame);

    rs2::points points = pointcloud.calculate(depth_frame);

    points_to_pcl(color_frame, points);

    auto cloud_map = cloudGroundRemover(info->cloud);
}
```

MartyG-RealSense, 2021-03-27:
Hi @idenc  In the small number of past 'tombstone' error cases on Android involving **Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR)** that have been reported, they usually end up unresolved.  Your mention of the error occurring no matter what you try is consistent with these past cases.  So if you have a working alternative (albeit a non-ideal one) then it may be best to focus on that.

Having said that, a rare successful fix for a SIGSEV code 1 error (though an error not necessarily caused by the same problem as your case) was developed in the case linked to below.

https://github.com/IntelRealSense/librealsense/issues/4214#issuecomment-506802514

In regard to using deep copy, advice provided in the link below may be helpful to you in deciding whether it is an appropriate method for your particular application.

https://github.com/IntelRealSense/librealsense/issues/3287

Given that your original script uses WaitForFrames() synchronously, I wonder whether that is causing a problem with trying to adapt the script for asynchronous operation as WaitForFrames() is a synchronous function, as described in this link:

https://github.com/IntelRealSense/librealsense/issues/3817#issuecomment-485663496

idenc, 2021-03-27:
I found which line causes the segfault, it is 
`pointcloud.map_to(color_frame);`

The first link is an issue for 32-bit devices, I am running on a 64-bit device so I don't think that is the issue, especially since it works when running synchronously.
 
I tried using frame_queue and keep without success. It is still not a deep copy as they still point to the same place in memory and it seems like there is no way to do a deep copy.

In terms of using a callback, the Java interface doesn't support using a callback with a pipeline

MartyG-RealSense, 2021-03-28:
The SDK's Java implementation contains a file called **FrameCallback.java**

http://docs.ros.org/en/kinetic/api/librealsense2/html/interfacecom_1_1intel_1_1realsense_1_1librealsense_1_1FrameCallback.html

My knowledge of Android programming in librealsense is admittedly limited.  My research did though find an example of the above file being imported into a librealsense Android project.  It can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/5520

[image]

closed 2021-04-09 by MartyG-RealSense