realsenseai/librealsense #8682: make: *** [Makefile:149: all] Error 2

Pruetikorn1224, 2021-03-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  d435 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                         | 
| Language                            |  python3.8                           | 
| Segment			|  Robot                           | 

### Issue Description
I try to install "pyrealsense2" and in the method "make -j4", I received this message:

[ 12%] Built target realsense-file
make: *** [Makefile:149: all] Error 2

What should I do to fix this?

MartyG-RealSense, 2021-03-28:
Hi @Pruetikorn1224   Are you installing on a desktop / laptop PC or on a computing device with an Arm processor chip such as Raspberry Pi or Nvidia Jetson, please?

Pruetikorn1224, 2021-03-28:
@MartyG-RealSense 
Currently, I am installing on laptop PC.

MartyG-RealSense, 2021-03-28:
Thank you very much.  Before we continue, it is worth highlighting that pyrealsense2 **pip** wheel files for Python 3.8 and 3.9 have just been added.  So if you are familiar with pip install, this could be an alternative installation option to a source-code build, using **pip install pyrealsense2**

https://pypi.org/project/pyrealsense2/#files

Pruetikorn1224, 2021-03-28:
Thank you! @MartyG-RealSense 

I used **pip3 install pyrealsense2** and it worked.

closed 2021-03-28 by Pruetikorn1224

MartyG-RealSense, 2021-03-28:
Great news about your success - thanks for the update  :)