realsenseai/librealsense #8683: ROS dependencies have changed

3dsf, 2021-03-28:
To enable the saving of ROS `.bag` files on linux via the manual installation, the dependencies have changed and the third party ROS package and should be updated appropriately.  

---  

ROS used to install with 
`sudo dnf install gcc-c++ python-rosdep python3-rosinstall_generator python-wstool python-rosinstall @buildsys-build`

Now it installs with 
`sudo dnf install gcc-c++ python3-rosdep python3-rosinstall_generator python3-vcstool @buildsys-build`  

---

the key package is `python3-vcstool` (aka [vcstool)](http://wiki.ros.org/vcstool) instead of `python-wstool` 

This has affected the **librealsense** install for **.01+** versions.  

---  

_I suspect it is the root for Windows users as well._

MartyG-RealSense, 2021-03-28:
Hi @3dsf   Thanks very much for your report.  I will tag @doronhi the RealSense ROS wrapper developer into this discussion to highlight your message.

MartyG-RealSense, 2021-04-03:
Hi @doronhi   Can you take a look at the question of @3dsf above please?  Thanks!

MartyG-RealSense, 2021-04-09:
@doronhi   Could you comment on the question in this case please?  Thanks very much!

3dsf, 2021-04-15:
I should probably clarify that this issue was experienced in **realsense-viewer**, thought I would suspect it would be problematic elsewhere.  

5.11.8-200.fc33.x86_64

After installing  `vcstool` the issues were resolved, thought I don't know if required a rebuild, but I probably did that.

MartyG-RealSense, 2021-04-16:
Thanks very much @3dsf for sharing the details of the solution that worked for you!

3dsf, 2021-04-16:
The issue still exists, though my install was working when I first made the post, and I was sharing the issue/solution.

I had been having issues with recording `.bag` files for a good while.  `.bag` files are ROS, so looked around to see if something had changed there recently, and then shared what I had found when it fixed my problem.

I mostly issued the update to clarify that this happened with **realsense-viewer**.

MartyG-RealSense, 2021-04-22:
Case closed due to solution achieved.

closed 2021-04-22 by MartyG-RealSense