realsenseai/librealsense #8689: Camera parameter acquisition D435i

987428377, 2021-03-29:
OK ，thank you read my problems！
I input command    
    roslaunch realsense2_camera rs_camera.launch
    rostopic echo /camera/color/camera_info
and then I got some imformation as follows:  

Camera.width: 640
Camera.height: 480
--------------------------------------------------------------------
header: 
  seq: 1009
  stamp: 
    secs: 1616939816
    nsecs:  37440538
  frame_id: "camera_color_optical_frame"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [607.324462890625, 0.0, 320.5830383300781, 0.0, 607.5777587890625, 241.2068328857422, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [607.324462890625, 0.0, 320.5830383300781, 0.0, 0.0, 607.5777587890625, 241.2068328857422, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---------------------------------------------
 and I need Camera.fx ; Camera.fy; Camera.cx; Camera.cy; Camera.k1; Camera.k2; Camera.p1; Camera.p2;  how can I get them.

MartyG-RealSense, 2021-03-29:
Hello, a simple method for retrieving the intrinsics of the camera is to use the '-c' calibration information command of the librealsense **rs-enumerate-devices** tool:

**rs-enumerate-devices -c**

When this command is input into the Windows command prompt or the Ubuntu terminal then a long list of the detected camera's calibration information is generated in the terminal window.  You can then scroll upwards from the bottom of the list until you find the information relating to a particular stream profile (for example, shown below is the Depth 848x480 profile from a D435i).

[image]

Please note from the above image that the **coefficients** on the 400 Series cameras are set to zero for reasons described in the link below.

https://github.com/IntelRealSense/librealsense/issues/1430#issuecomment-375945916

These five distortion coefficients are **k1, k2, p1, p2, k3**.

More information about these intrinsic values can be found at the following documentation links:

https://intelrealsense.github.io/librealsense/doxygen/structrs2__intrinsics.html

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#intrinsic-camera-parameters

ppx and ppy are equivalent to the cx and cy values that you are seeking as both sets of terms describe the **principal point**.

Whilst rs-enumerate-devices provides an easy way to retrieve the calibration information, you can also do so with program scripting, as described in the discussion linked to below:

https://github.com/IntelRealSense/librealsense/issues/2930

987428377, 2021-03-29:
> When this command is input into the Windows command prompt or the Ubuntu terminal then a long list of the detected camera's calibration information is generated in the terminal window. You can then [...]

Thank you very much for your help！

closed 2021-04-06 by 987428377