realsenseai/librealsense #8691: Frame didn't arrive within 5000

Pruetikorn1224, 2021-03-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | N/A | 
| Operating System & Version |    Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.8.0                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.0                         | 
| Language                            |  python3.8                          | 
| Segment			|  Laptop                          | 

### Issue Description
Currently, I'm working on project on realsense d435i.

import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

pipeline.start(config)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Show images
        cv2.namedWindow('RealSense Depth', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense Depth', color_image)
        key = cv2.waitKey(1)
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break

except Exception as e:

    print(e)
    print('Device: ' + str(device))
    print('Product Line: ' + device_product_line)

finally:

    # Stop streaming
    pipeline.stop()

and I receive this error.

Frame didn't arrive within 5000
Device: <pyrealsense2.device: Intel RealSense D435I (S/N: 046222071432)>
Product Line: D400

How to fix this?

MartyG-RealSense, 2021-03-29:
Hi @Pruetikorn1224   This is an error where the code can be okay but the error occurs because of the particular computer that it is installed on, whilst the code may work fine on a different computer.

If you do not have another computer to try it on and you built the wrapper from source code on Python 3.8, there is now the option to build the wrapper for Python 3.8 and 3.9 using the **pip install pyrealsense2** instruction.

https://github.com/IntelRealSense/librealsense/issues/8686

Pruetikorn1224, 2021-03-29:
@MartyG-RealSense Ok, so I will change my computer to test on.

Thank you

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.