realsenseai/librealsense #8692: algorithm topic about Output camera pose with d435i

987428377, 2021-03-29:
Thank you for your answers and support! I can already run rtabslam with d435i. So what function should I use to get the position and attitude of camera in modeling? Does he have any existing algorithm topic

MartyG-RealSense, 2021-03-29:
Hello, the D435i does not have built-in support for pose data (position and angle).  There are a range of ways to obtain pose of the camera or of an observed object though.  It sounds as though you wish to retrieve the pose of the **camera**.

Intel have a ROS guide for using D435i with RTABMAP for the purposes of SLAM.  A RealSense user expanded upon that guide to add the ability to retrieve the camera's **relative position** from the generated map by performing **localization**.  The guide describes that the pose can then be published to ROS.

https://github.com/IntelRealSense/librealsense/issues/8094#issuecomment-753979162

That discussion is also worth reading from the start for information about other methods for obtaining camera pose.

The subject of localization for D435i is discussed further here:

https://github.com/IntelRealSense/realsense-ros/issues/1529#issuecomment-735299914

987428377, 2021-03-29:
> Hello, the D435i does not have built-in support for pose data (position and angle). There are a range of ways to obtain pose of the camera or of an observed object though. It sounds as though you wish to retrieve the pose of the **camera**.
> 
> Intel have a ROS guide for using D435i with RTABMAP for the purposes of SLAM. A RealSense user expanded upon that guide to describe how to retrieve the camera's **relative position** from the generated map by performing **localization**. The guide describes that the pose can then be published to ROS.
> 
> [#8094 (comment)](https://github.com/IntelRealSense/librealsense/issues/8094#issuecomment-753979162)
> 
> That discussion is also worth reading from the start for information about other methods for obtaining camera pose.
> 
> The subject of localization for D435i is discussed further here:
> 
> [IntelRealSense/realsense-ros#1529 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/1529#issuecomment-735299914)
> 
> If you are using **RTABSLAM** and would prefer a pose solution based around the system that you are already using, the RTABSLAM discussion about pose in the link below may be helpful.
> 
> [introlab/find-object#23](https://github.com/introlab/find-object/issues/23)
Thank you for you reply!

MartyG-RealSense, 2021-04-06:
hi @987428377  Do you require further assistance with this case, please?  Thanks!

987428377, 2021-04-10:
> hi @987428377 Do you require further assistance with this case, please? Thank you very much !

Thanks！ I will try this method http://docs.px4.io/master/en/computer_vision/visual_inertial_odometry.html

closed 2021-04-10 by 987428377