realsenseai/librealsense #8694: Multi camera set up without hw sync, endless loop on some cameras

Hunk86, 2021-03-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | 3 times D435, 1 D455  | 
| Firmware Version                   | 05.12.11.00 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.4.0-70-lowlatency                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.42                           | 
| Language                            |  C++ / ROS                          | 
| Segment			|  Robot                  | 

### Setup
1. Every Camera is connected directly to the computer to a USB 3 connector
2. I've installed all librealsense debian packages
3. Udev are setup
```
cat /etc/udev/rules.d/98-realsense-power.rules 
# This disables autosuspend and auto-power-off of the usb2 ports when a Realsense camera is connected
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1d6b", ATTRS{idProduct}=="0003", TEST=="../power/control", ATTR{../power/control}="on"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1d6b", ATTRS{idProduct}=="0003", TEST=="../power/autosuspend", ATTR{../power/autosuspend}="-1" 
SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b3a", TEST=="../power/autosuspend", ATTR{../power/autosuspend}="-1" 
```

### Application

I start the realse node four times with :

`roslaunch realsense2_camera rs_camera.launch ...`

and the important args ( serial ... )

The main problem is that the cameras do not boot consistently when I start them at the same time. After various restarts of the individual non-functioning nodes, I get the point clouds from all cameras.

I've added different issues descriptions in the bottom. If I can provide you more information please ask for it :)

Thank you for your help!

### Issue Description

1. Infinite loop in node

```
[ INFO] [1617009159.138296257]: Device with serial number 043422251243 was found.
[ INFO] [1617009159.138319818]: Device with physical ID 2-1-2 was found.
[ INFO] [1617009159.138328112]: Device with name Intel RealSense D455 was found.
[ INFO] [1617009159.138566633]: Device with port number 2-1 was found.
 29/03 09:12:39,154 ERROR [140437490169600] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 29/03 09:12:39,154 ERROR [140437448206080] (sensor.cpp:523) acquire_power failed: failed to set power state
 29/03 09:12:39,154 WARNING [140437448206080] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1617009159.170030048]: Device 3/4 failed with exception: failed to set power state
 29/03 09:12:39,169 ERROR [140437473384192] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 29/03 09:12:39,169 ERROR [140437448206080] (sensor.cpp:523) acquire_power failed: failed to set power state
 29/03 09:12:39,170 WARNING [140437448206080] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1617009159.193384440]: Device 4/4 failed with exception: failed to set power state
[ERROR] [1617009159.193410131]: The requested device with serial number 948122072273 is NOT found. Will Try again.
 29/03 09:12:39,192 ERROR [140437464991488] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 29/03 09:12:39,192 ERROR [140437448206080] (sensor.cpp:523) acquire_power failed: failed to set power state
 29/03 09:12:39,193 WARNING [140437448206080] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1617009165.358216528]: Device 2/4 failed with exception: failed to set power state
[ INFO] [1617009165.220363650]:  
[ INFO] [1617009165.343130113]: Device with serial number 043422251243 was found.
[ INFO] [1617009165.343152383]: Device with physical ID 2-1-2 was found.
[ INFO] [1617009165.343159515]: Device with name Intel RealSense D455 was found.
[ INFO] [1617009165.343382924]: Device with port number 2-1 was found.
 29/03 09:12:45,357 ERROR [140437498562304] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 29/03 09:12:45,357 ERROR [140437448206080] (sensor.cpp:523) acquire_power failed: failed to set power state
 29/03 09:12:45,358 WARNING [140437448206080] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1617009165.373676648]: Device 3/4 failed with exception: failed to set power state
 29/03 09:12:45,373 ERROR [140437481776896] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 29/03 09:12:45,373 ERROR [140437448206080] (sensor.cpp:523) acquire_power failed: failed to set power state
 29/03 09:12:45,373 WARNING [140437448206080] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1617009165.399614684]: Device 4/4 failed with exception: failed to set power state
[ERROR] [1617009165.399642243]: The requested device with serial number 948122072273 is NOT found. Will Try again.
 29/03 09:12:45,399 ERROR [140437473384192] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 29/03 09:12:45,399 ERROR [140437448206080] (sensor.cpp:523) acquire_power failed: failed to set power state
 29/03 09:12:45,399 WARNING [140437448206080] (rs.cpp:306) null pointer passed for argument "device"
```

But when I stop it and run it again and program runs without any problems. 

My guess here is that the camera drivers are interfering with simultaneous startup.

2. Feedback when running 
```
 29/03 09:13:05,371 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 29/03 09:13:05,421 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1617009185.478911989]: publishing unordered pointcloud.
 29/03 09:13:05,572 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:05,676 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:05,726 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:05,776 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:06,028 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:06,179 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:06,229 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:06,379 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:06,430 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:06,580 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:06,631 WARNING [140694766860032] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/03 09:13:07,196 ERROR [140693833201408] (ds5-options.cpp:86) Asic Temperature value is not valid!
 29/03 09:13:08,197 ERROR [140693833201408] (ds5-options.cpp:86) Asic Temperature value is not valid!
 29/03 09:13:09,197 ERROR [140693833201408] (ds5-options.cpp:86) Asic Temperature value is not valid!
```

When the node is running, I get this libusb feedback at the beginning.

3. rs-enumerate-devices

```
Could not create device - failed to set power state . Check SDK logs for details
 29/03 13:02:28,933 ERROR [140253063001856] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 29/03 13:02:28,933 ERROR [140253270798528] (sensor.cpp:523) acquire_power failed: failed to set power state
Could not create device - failed to set power state . Check SDK logs for details
```

Here I also get this feedback similar to the node

4. UDev Rule

One more question about the udev rule comment. It says that auto power off is done for usb2. Does this also apply to USB3?

MartyG-RealSense, 2021-03-29:
Hi @Hunk86   Have you seen Intel's guide for setting up multiple cameras in ROS on the same computer and obtaining point clouds from them, please?

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

Hunk86, 2021-03-29:
@MartyG-RealSense 
yes I've seen this, but this isn't my problem.

The example above is an easy example for two cameras.  I've problem with four cameras with the librealsense software driver layer

MartyG-RealSense, 2021-03-29:
The version of this guide on the official Intel RealSense website adds further sections that the original GitHub guide does not have, including guidance on adding further cameras on the same computer.  It can be found around halfway down the guide in the section headed **Adding More Cameras**.

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

This section comments on how adding further cameras to the same computer can place additional demands on the computer and USB, and this is true.  A seminar about multiple cameras held by Intel in 2018 suggested a computer with an Intel Core i7 processor for a system with 4 cameras attached.

If you have a computer with a high specification then it may be worth adding cameras in the method in the above expanded guide if you are not implementing a 4-camera setup that way already.

So long as each camera is roslaunched on its own individual ROS terminal as indicated by the guide then as far as I know there should not be an activation conflict between the multiple cameras.

If the **control_transfer returned** error is continuously generating then this indicates that there is a serious communication problem with the device (which would be consistent with the **Failed to set power state** errors).

If you are launching each camera in its own terminal, a good place to begin the investigation of your case would be to add the command below to the end of all of your roslaunch statements to reset the camera at launch:

**initial_reset:=true**

MartyG-RealSense, 2021-04-06:
Hi @Hunk86  Do you require further assistance with this case, please?  Thanks!

Hunk86, 2021-04-09:
@MartyG-RealSense 

sorry for the late response

We tried the inital_reset:=true but I've exactly the same problems.

Also every launch file is running in its own terminal. 

Do you have more ideas what we can test or how we can provide you more information for debugging ?

MartyG-RealSense, 2021-04-09:
An alternative to launching specific devices using serial number is to use **usb_port_id** instead of **serial_number** in the roslaunch instruction.  This method looks for the camera on a USB port of the computer with a specific ID.  For example, adding this cimmand to the roslaunch instruction to look for the camera on a USB port with the ID '4-1':

**usb_port_id:=4-1**

MartyG-RealSense, 2021-04-21:
Case closed due to no further comments received.

norman1479, 2021-05-25:
Hi, please re-open this issue,

I also work with the setup described above and have switched to usb_port_id for identifcation, but we still get `RS2_USB_STATUS_BUSY`/`failed to set power state` errors.

Nodes do not always catch this, when launching into the multi-camera setup we always find one or two of the nodes shutting down with an exception due to the same error. Sometimes, we even have to reboot or re-power the computer, or re-connect the cameras via the kernel driver to make the failed cameras detectable again.

MartyG-RealSense, 2021-05-25:
Hi @norman1479  Given the similarity between your situation and that of @Hunk86 who created the case, I will re-open the case for further discussion.

@Hunk86   Please feel free to contribute if you wish to.  Thanks!

Hunk86, 2021-05-27:
Hello  @MartyG-RealSense 

thank you for reopen the Issue

I still have the same problems, right now I've the case that one camera can't be connected, but the other ones are working fine

`[ INFO] [1622101306.275143735]:  

 27/05 07:41:46,286 ERROR [140318594823936] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY

 27/05 07:41:46,286 ERROR [140318951261952] (sensor.cpp:526) acquire_power failed: failed to set power state

 27/05 07:41:46,287 WARNING [140318951261952] (rs.cpp:306) null pointer passed for argument "device"

 27/05 07:41:46,302 ERROR [140318229255936] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY

[ WARN] [1622101306.303164557]: Device 2/4 failed with exception: failed to set power state

 27/05 07:41:46,302 ERROR [140318951261952] (sensor.cpp:526) acquire_power failed: failed to set power state

 27/05 07:41:46,303 WARNING [140318951261952] (rs.cpp:306) null pointer passed for argument "device"

[ WARN] [1622101306.314305245]: Device 3/4 failed with exception: failed to set power state

 27/05 07:41:46,313 ERROR [140318237648640] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY

 27/05 07:41:46,314 ERROR [140318951261952] (sensor.cpp:526) acquire_power failed: failed to set power state

 27/05 07:41:46,314 WARNING [140318951261952] (rs.cpp:306) null pointer passed for argument "device"

[ WARN] [1622101306.326714277]: Device 4/4 failed with exception: usb device disconnected

[ERROR] [1622101306.326741365]: The requested device with usb port id 2-4 is NOT found. Will Try again.

 27/05 07:41:46,326 WARNING [140318594823936] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 135 error: No such device, number: 19

 27/05 07:41:46,326 WARNING [140318594823936] (uvc-device.cpp:183) rs_uvc_device exception in listen_to_interrupts method: failed to submit interrupt request, error: RS2_USB_STATUS_OTHER

 27/05 07:41:46,326 WARNING [140318594823936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
`

after unplug and plug again, it was possible and I got only this kind of errors

`
 27/05 07:45:35,775 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

[ WARN] [1622101535.842774237]: Could not load crop_box_negative_points, using default value

[ WARN] [1622101535.842821131]: Can't add crop_box_negative filter because of invalid number of crop_box_negative_points:[]

[ WARN] [1622101535.960479156]: 

[ INFO] [1622101535.825982943]: SELECTED BASE:Depth, 0

[ INFO] [1622101535.832291259]: RealSense Node Is Up!

[ INFO] [1622101535.842244003]: Added Filter crop_box_positive

[ INFO] [1622101535.843534617]: Added Filter approximate_voxel

[ INFO] [1622101535.844181354]: Added Filter radius_outlier_removal

[ INFO] [1622101535.848621680]: Added Filter approximate_voxel

[ INFO] [1622101535.849294409]: Added Filter radius_outlier_removal

 27/05 07:45:35,964 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

 27/05 07:45:36,014 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

 27/05 07:45:36,065 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

 27/05 07:45:36,115 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

[ INFO] [1622101536.166777391]: publishing unordered pointcloud.

 27/05 07:45:36,289 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

 27/05 07:45:36,339 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

 27/05 07:45:36,389 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

[ WARN] [1622101536.718460762]: Hardware Notification:Right MIPI error,1.6221e+12,Error,Hardware Error

 27/05 07:45:37,106 ERROR [140180594841344] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130

[ WARN] [1622101537.718730376]: Hardware Notification:Right MIPI error,1.6221e+12,Error,Hardware Error

 27/05 07:45:38,343 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

 27/05 07:45:38,393 WARNING [140181222979328] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61`

MartyG-RealSense, 2021-05-27:
Hi @Hunk86  Is the problem always with the same camera wit a specific serial number?  Or is it that if you have a set of cameras attached, one of the cameras in the set will always have a **Failed to set power state** - but not always the exact same camera.

Hunk86, 2021-05-27:
Hello @MartyG-RealSense  it is random and not always the same. We have cases we we get all cameras up and sometimes only one 

We are also using dev_id right now and not serial number

Do you have any idea ?

MartyG-RealSense, 2021-05-27:
There was a past case in which a **right MIPI error** occurred because of frames not arriving reliably.  This would be consistent with the accompanying **control_transfer returned** warning in your log, which can indicate that there is a serious communication problem with the device if the message is generating continuously instead of appearing a small number of times at launch.

As the issue is with a randomly selected camera rather than the same camera each time, it suggests that the problem is not with the camera hardware or its firmware.

The discussion in the link below may be useful, which has a similar RealSense ROS scenario of some cameras failing at start-up sometimes but not every time.  It discusses a range of possible solutions, such as putting a sleep period between the start of each camera.

https://github.com/IntelRealSense/realsense-ros/issues/1187

At the end of that case, **Doronhi** the RealSense ROS wrapper developer provides advice about occurrences of **RS2_USB_STATUS_BUSY**

https://github.com/IntelRealSense/realsense-ros/issues/1187#issuecomment-759367599

norman1479, 2021-05-27:
Additional errors we frequently encountered in this scenario are RS2_USB_STATUS_NOT_FOUND and  RS2_USB_STATUS_NO_DEVICE.

In this case, one or more cameras (or the corresponding usb port / driver) remain in an undetactable state we can only recover by repowering the computer or issue a USBDEVFS_RESET using ioctl on that port - even after re-plugging the camera or after an unbind/bind of the device via /sys/bus/usb... and also when using initial_reset:=true. 

That's what our kern.log repeats in that situation for the camera port related: `usb 2-2: usbfs: usb_submit_urb returned -19`

We will also try the suggestions mentioned above and try a build with v4l next.

MartyG-RealSense, 2021-05-27:
Thanks very much @norman1479 for the update on your particular situation - good luck with your tests and I look forward to your update afterwards.

Hunk86, 2021-05-27:
@MartyG-RealSense 

When I check the documentation

option(FORCE_LIBUVC "Explicitly turn-on libuvc backend - deprecated, use FORCE_RSUSB_BACKEND instead" OFF)<br class="Apple-interchange-newline">

The LIBUVC is deprecated ? But I will try to turn this on and disable the backend

MartyG-RealSense, 2021-05-27:
The RSUSB backend system was introduced in SDK **2.30.0**.  Before that, libuvc-backend was typically used to perform an installation over an internet connection without the need for kernel patching.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc-backend method, though deprecated, is still valid as far as I know, though rarely used now.

MartyG-RealSense, 2021-06-02:
Hi @Hunk86  Do you have an update about this case that you can provide, please?  Thanks!

Hunk86, 2021-06-05:
Yes at the moment we try with the new backend but we still have the same problems with multiple cameras

I can provide you some logs next week

MartyG-RealSense, 2021-06-05:
Thanks very much @Hunk86 - I look forward to your update in the coming week.  Good luck!

closed 2021-06-18 by MartyG-RealSense