realsenseai/librealsense #8695: Capturing 1 or n number of frames for image processing.

PlasticBlaze, 2021-03-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi 4|
| SDK Version                            | 2.43.0                          | 
| Language                            |  opencv/python }                          | 
| Segment			|  Robot                       | 

### Issue Description
Hello,
Somewhat beginner here, my project is going to be running a routine, travel a distance, capture and process image(s) to give coordinate feedback, perform action. Repeat process. The surface it will be capturing is the ground so everything will be stationary and the objects it will be finding will be stationary.
My question is, is there a better way to capture a single frame or a maybe a group of frames that can be averaged/ merged (cv2.addedweight opencv) and further used for image processing. I have written a basic code following the examples and using cv2.imwrite as follows below. This does work and saves a "1.jpg" image file to the file directory that I can eventually cv2.imread and run the image processing and coordinate functions on. However, I pretty sure this is not the best or memory efficient way to achieve what I am trying to get. I figured someone else knows a better method to achieve this.

```
import pyrealsense2 as rs
import numpy as np
import cv2
import json

json1 = json.load(open("json_string.json"))
# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()

config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)

#config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
cfg = pipeline.start(config)
device = cfg.get_device()

json1 = json.load(open("json_string.json"))
json_string= str(json1).replace("'", '\"')
advanced_mode = rs.rs400_advanced_mode(device)
advanced_mode.load_json(json_string)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()

        # Convert images to numpy arrays
        colorizer = rs.colorizer()
        depth_image = np.asanyarray(colorizer.colorize(depth_frame).get_data())
        cv2.imwrite('/home/pi/Pictures/1.jpg', depth_image)
        img5 = cv2.imread('/home/pi/Pictures/1.jpg', cv2.IMREAD_COLOR)

        #color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        #depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.08), cv2.COLORMAP_JET)

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', depth_image)
        cv2.waitKey(1)

finally:

    # Stop streaming
    pipeline.stop()
```

MartyG-RealSense, 2021-03-30:
Hi @PlasticBlaze  I am not aware of a Python code example for the **Keep()** function that demonstrates how to apply post-processing to the stored frames  There is such an example for C++ though that may provide useful insights about structuring an equivalent process in Python.

https://github.com/IntelRealSense/librealsense/issues/1942#issuecomment-400031901

If you are only saving a single frame or a small collection of frames at a moment in time when a particular distance on the travel route is reached though then using Keep() may not provide a significant advantage compared to writing the frames as image files instead of storing them in memory.

In regard to possible alternative methods to Python, the link below discusses a RealSense project with similarities to yours that took the approach of using **ROS**.  The camera was mounted on a moveable module on a linear rail that would move along the rail capturing an image of the ground at specific defined positions on the route.

https://github.com/IntelRealSense/realsense-ros/issues/1581

PlasticBlaze, 2021-03-30:
I'm not sure if keep() is the right thing to use for me. I'm just not knowledgeable enough at the moment to figure out how else to do it with OpenCV or anything else. I know there are things like cv2.videocapture that I might be able to use. Can you leave this open to see if anyone that does know OpenCV fairly well might be able to answer?

MartyG-RealSense, 2021-03-30:
A typical use of Keep() would be to store a short recording sequence (such as 30 seconds maximum) in the memory of the computer and then batch-process the frames all at the same time at the end of the recording process (e.g apply post-processing, align depth to color and save the frames).  It can be useful in situations where there may be lag in the recording process when writing the frames to storage space continuously during recording.

Keep() is suited to short recording durations because the computer's memory capacity quickly becomes consumed by the frames being stored in it.

Yes, it is fine to keep this case open for a further time period.

PlasticBlaze, 2021-04-07:
Hey Marty, I haven't had a chance to try this yet but I had still been looking for a solution and came across another github page where a guy did a project with GPS, realsense camera, and image geo-tracking. One of his .py files actually looks to save 100 frames from just looking at it. I can try it tonight to see. 

This .py is [here](https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/single_frameset.py)

and the full project is [here](https://github.com/soarwing52/RealsensePython). This project shows a ton of great information and ideas for beginners like myself and others.

MartyG-RealSense, 2021-04-07:
Thanks so much @PlasticBlaze  for sharing the resources!   Please do update when you have had an opportunity to test.  Good luck!

PlasticBlaze, 2021-04-08:
Hey Marty,
I tried that small snippet of code and it doesn't seem to do anything. After doing even more digging around I actually found where the same user was actually asking in the past about this method. It was number [#4020](https://github.com/IntelRealSense/librealsense/issues/4020)

MartyG-RealSense, 2021-04-08:
I remember the project of **soarwing52** referenced in https://github.com/IntelRealSense/librealsense/issues/4020 - I believe that the final version of the **save_single_frameset** code that soarwing52 was referring to in that case can be found on their GitHub's folder of standalone Python scripts for various functions of their project.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/single_frameset.py

The full list of standalone functions created by soarwing52 - including a Python conversion of the SDK's C++ example **rs-measure** for measuring between two points - can be found here:

https://github.com/soarwing52/RealsensePython/tree/master/separate%20functions

closed 2021-04-21 by MartyG-RealSense