realsenseai/librealsense #8697: Is it possible to refer to the distance of a specific coordinate in a color image if the resolution of the color image and the depth image are different?

nkanbe, 2021-03-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |1.5.5.0 | 
| Operating System & Version |   Ubuntu18 | 
| Kernel Version (Linux Only)    | 4.9.140-tegra | 
| Platform                                 | NVIDIA Jetson NX|
| SDK Version                            |  | 
| Language                            | Python,Opencv| 

### Issue Description
Hi
I am currently using JetsonNX, RealsenseL515 and pyreansense2.

Previously, I used the D435i to fix the distortion and did the following:
(The resolution of color images and depth images is the same)
1, Detect a person in a color image and get the coordinates
2, get the distance from the depth of the same coordinates 

However, in L515, the resolutions of the color image and the depth image do not seem to match.
https://github.com/IntelRealSense/librealsense/issues/6684

The current resolution is 1280 x 720 in color and 1024 x 768 in depth.
Therefore, when you run processes 1 and 2, they will see different positions because they have different coordinates.

How can I see the distance of the location shown in a color image if the resolutions are different? 

Thank you.

RealSenseSupport, 2021-04-02:
Hi @nkanbe 

You are right the color and depth for the L515 are in different resolutions.  You can have a look at our examples that take into consideration how to align the two streams.  The rs-align (https://github.com/IntelRealSense/librealsense/tree/master/examples/align) and rs-align-advanced (https://github.com/IntelRealSense/librealsense/tree/master/examples/align-advanced)

There are no plans to have common resolutions between the depth stream and the color stream.

nkanbe, 2021-04-07:
Thank you.
I found that the resolutions do not match

closed 2021-04-07 by nkanbe