realsenseai/librealsense #8698: How to set intensity setpoint in auto-exposure mode using c++?

fwf-lernen, 2021-03-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | 05.12.11.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0 | 
| Platform                                 | PC |
| SDK Version                            | 2.42.0 | 
| Language                            |  C | 

### Issue Description
I have found how to set intensity setpoint in auto-exposure mode using python at [here](https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras#section-2-4-manual-vs-auto-exposure).
``
cfg = rs.config()
cfg.enable_stream(rs.stream.infrared, 1) # For D400 infrared streams are split between 1 and 2, for left and right

p = rs.pipeline()
profile = p.start(cfg)
advnc_mode = rs.rs400_advanced_mode(profile.get_device())

\# Read-modify-write of the AE control table
ae_ctrl = advnc_mode.get_ae_control()
ae_ctrl.meanIntensitySetPoint = 1000
advnc_mode.set_ae_control(ae_ctrl)

\# After modifying the setpoint its recommended to wait for several frames to make sure the change takes effect
for j in range(1, 10):
    p.wait_for_frames()
``
But I want to know how to rewrite it in c++.

fwf-lernen, 2021-03-30:
I think I have solve the problem.
`
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, imgWidth, imgHeight, RS2_FORMAT_Z16, 15);
    cfg.enable_stream(RS2_STREAM_INFRARED, 1, imgWidth, imgHeight, RS2_FORMAT_Y8, 15);
    cfg.enable_stream(RS2_STREAM_INFRARED, 2, imgWidth, imgHeight, RS2_FORMAT_Y8, 15);
    rs2::pipeline_profile selection = pipe.start(cfg);
    rs2::device selected_device = selection.get_device();
    rs400::advanced_mode advnc_mode(selected_device);
    STAEControl ae_ctrl;
    ae_ctrl.meanIntensitySetPoint = 1500;
    advnc_mode.set_ae_control(ae_ctrl);
`

MartyG-RealSense, 2021-03-30:
Hi @fwf-lernen   Thanks very much for sharing your setpoint-defining Python code!

I was piecing together a C++ conversion of your script, so I'm very pleased that you achieved it first.  :)

fwf-lernen, 2021-03-30:
Hi @MartyG-RealSense You are welcome.
The setpoint-defining Python code is found link below:
https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras#section-2-4-manual-vs-auto-exposure
I just convert it to c++.

closed 2021-03-30 by fwf-lernen