realsenseai/librealsense #8699: How do I get the rotation of L515?

jack-wlj, 2021-03-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) / Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/ NVIDIA Jetson NX |
| SDK Version                            |  { legacy / 2.41.0 }                          | 
| Language                            |  {C/C++/opencv }                          | 
| Segment			|  {UE4/VR/AR/others }                          | 

### Issue Description
Hi,
    I want to get the rotation of the L515, what do I need to do?Only one L515 will be used?
    I want to get the L515 rotation data like the T265 rotation data?I have tried some methods, but nothing works. Is there any way?
    What do I need to do to fulfill my needs?
    Looking forward to your effective reply. Thank you very much!

RealSenseSupport, 2021-04-02:
Hi @jack-wlj 

So you are looking for a SLAM solution using the L515 camera instead of just using the raw IMU data that is provided from the L515 camera.  The T265 was purpose built for SLAM tracking with it's core algorithm.  The L515 does not have native SLAM capability which means that a 3rd party solution would need to be investigated.  Not familiar with 3rd party SLAM for L515.

jack-wlj, 2021-04-06:
> 嗨@ jack-wlj
> 
> 因此，您正在寻找使用L515摄像机而不是仅使用L515摄像机提供的原始IMU数据的SLAM解决方案。T265的核心算法专门用于SLAM跟踪。L515不具有本地SLAM功能，这意味着需要研究第三方解决方案。不熟悉L515的第三方SLAM。

Thanks for your reply!I don't use SLAM, I just want to get the rotation of L515 itself, such as looking down at 45° manually,How do I get the value of 45°?
Since the depth data of L515 is highly accurate, I want to use the depth value and its own rotation to calculate the height of L515 from the ground, so I need the L515's own rotation. I hope you can help me solve this problem. Thank you again.

RealSenseSupport, 2021-04-14:
Hi @jack-wlj 

There is nothing specific in our SDK that provides this information for the L515.  You may have to look at a few factors, the actual extrinsic of the camera, meaning the mechanical information of the camera itself that is outlined in the datasheet and the CAD model

https://dev.intelrealsense.com/docs/lidar-l500

You may also be able to use the intrinsic and extrinsic calibration information provided by the APIs, an example in getting this information is in the rs-enumerate-devices sample - https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

Hope this helps

jack-wlj, 2021-04-17:
> Hi @jack-wlj
>
> There is nothing specific in our SDK that provides this information for the L515. You may have to look at a few factors, the actual extrinsic of the camera, meaning the mechanical [...]

I have read all the information, but it is a pity that there is no information I need. I don't understand why L515 can't get quaternions？

RealSenseSupport, 2021-05-07:
Hi @jack-wlj 

Yes sorry to say that L515 does not generate quaternion values.  The only RealSense device that we have that provides such data from the device itself is the T265 tracking camera.

closed 2021-05-20 by jack-wlj