realsenseai/librealsense #8701: L515 no detection

hyoon970, 2021-03-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | v2.42.2 | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC, Jupyter notebook  |
| SDK Version                            |  2.0 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello,

I'm using L515, and it is detected on Intel RealSense viewer.

However, when I use python code for recording, it says
------------------------------------------------------
File "record2.py", line 63, in <module>
    profile = pipeline.start(config)
RuntimeError: No device connected
-------------------------------------------------------

The code record2.py is like below.
-------------------------------------------------------------
pipeline = rs.pipeline()
    config = rs.config()
    #config.enable_device(f0190164)
    config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
    # Start pipeline
    profile = pipeline.start(config)
    frames = pipeline.wait_for_frames()
    color_frame = frames.get_color_frame()
--------------------------------------------------------------

Depth and color option on the code is same as my camera.

How can I solve this problem?

Thanks and Regards,

H.Yoon

RealSenseSupport, 2021-04-02:
Hi @hyoon970 

I'm not able to replicate what you are seeing on my current setup.  I'm using Python2.7 on my machine.  I do not see the no device detected error.  I'm also using the built files from the 2.43 Librealsense SDK Windows install package.

Can you add images of what you're seeing.  I also cut and pasted your code in one of our python examples that way I know the code was added to an example that was previously ran without errors.

hyoon970, 2021-04-08:
Thank you for your reply.

I solved the problem and I can use L515 now.

RealSenseSupport, 2021-04-09:
Excellent

closed 2021-04-09 by RealSenseSupport