realsenseai/librealsense #8704: Not all frames run?

ls880, 2021-03-30:
When running functions like pointcloud and rosbag_example on matlab, I'm unsure these run every frame. If I stop at the same point running two different times this gives a different output? How do I ensure every frame is chosen to run?

MartyG-RealSense, 2021-03-30:
Hi @ls880   A RealSense team member provides guidance in the link below about how to obtain the timestamp of each frame of a program in the RealSense MATLAB wrapper, such as **rosbag_example.m**.  Is this helpful to you, please?

https://github.com/IntelRealSense/librealsense/issues/3965#issuecomment-494119309

ls880, 2021-04-05:
Sorry for the late reply, I have been having computer problems. I have tried this and I get an error stating that timestamp domain is the system time, not the frame time. This also come up on the RS viewer tool. How do I change this?

ls880, 2021-04-05:
It also states that per frame metadata isn't enable at OS level and to follow installation guide for details

ls880, 2021-04-05:
But essentially I still have the problem that the frames do not run in order? Even using this feature I can see that the frames seemingly come out randomly?

MartyG-RealSense, 2021-04-05:
If you are using the RealSense MATLAB wrapper then you are likely using Windows 10 on your computer.  Is that correct, please?

If you are using Windows, then you can enable metadata for your camera in the **RealSense Viewer** program.   If a **Frame Metadata Disabled** box pops up in the corner of the Viewer when it is launched then left-click on the box's **Enable** button to enable per-frame metadata.

[image]

ls880, 2021-04-06:
Hi, I have tried this but when I run the code files the frames still do no run in order, they seem completely random

MartyG-RealSense, 2021-04-07:
A bag file always runs in the forward direction when played and does not run backwards.  So there is a certain amount of predictable linearity in the playback, and once the bag reaches the end then it will loop around to the start and play from start frame to end again unless playback looping is disabled.

 Could you tell me please which method you use to stop the bag playback and check the frame number?  If you were using a 'stop' button then it would be very hard to stop the playback on the same frame number each time  due to te speed at which frames are played back.

ls880, 2021-04-12:
I am currently using the beginning of the pointcloud example to run through the frames and have put a break point at the end of the loop. Many frames seem to be missing when using this method. I have been advised that with .bag files, some frames are not able to be captured fast enough and the next one will be used instead. Is there a way I can ensure that all frames are captured?

function pointcloud_example(Cross)
    % Make Config object to manage pipeline settings
    % From rosbag_example
    cfg = realsense.config();
    validateattributes(Cross, {'char','string'}, {'scalartext', 'nonempty'}, '', 'Cross', 1);
    % Tell pipeline to stream from the given rosbag file
    cfg.enable_device_from_file(Cross)

    % Make Pipeline object to manage streaming
    pipe = realsense.pipeline();
    
    pointcloud = realsense.pointcloud();

    % Start streaming from file with default settings
    profile = pipe.start(cfg);
    
    figure('units','normalized','outerposition',[0 0 1 1])

    % Main loop
    for i = 1:600

        % Obtain frames from a streaming device
        fs = pipe.wait_for_frames();

ls880, 2021-04-12:
Is it possible o implement the playback.m rather than pipeline.m and alter the playback speed, line 49?

MartyG-RealSense, 2021-04-13:
Apologies for the delay in responding further.  Looking through your code, I notice that it is not using a command called **playback.set_real_time(false);**

Setting **set_real_time** to False is commonly done to help ensure smooth playback when reading frames from a bag file.  A C++ example script that demonstrates this function is in the link below.

https://github.com/IntelRealSense/librealsense/issues/2743

[image]

ls880, 2021-04-13:
I have tried adding this in but I only seem to get error messages that there are not enough inputs, or "Unable to resolve the name playback.set_real_time"? I tried to implement similar to this issue: https://github.com/IntelRealSense/librealsense/issues/3126

cfg.enable_device_from_file(filename)

% Make Pipeline object to manage streaming
pipe = realsense.pipeline();

% Start streaming from the file with default settings
profile = pipe.start(cfg);
device   = profile.get_device();
run =  realsense.playback(device)
playback.set_real_time(false)

MartyG-RealSense, 2021-04-13:
Comparing your code to the linked-to script, it looks as though **playback** has not been defined in the line before it is called by playback.set_real_time

**rs2::playback playback = device.as<rs2::playback>();**
playback.set_real_time(false);

I also note that your quoted script above is missing some semicolons on the end of the lines, in these lines:

cfg.enable_device_from_file(filename)
run = realsense.playback(device)
playback.set_real_time(false)

ls880, 2021-04-13:
This is not valid code in matlab, so I wrote it as below but I still get the error. It seems to want a second input for the index?

Error:
Not enough input arguments.

Error in realsense.playback (line 6)
           this = this@realsense.device(handle, index);

Code:
profile = pipe.start(cfg);
device = profile.get_device();
playback = realsense.playback(device);
playback.set_real_time(false)

MartyG-RealSense, 2021-04-14:
I found an example of a RealSense user who wrote a **set_real_time** playback routine in the MATLAB wrapper and how it was debugged.

https://github.com/IntelRealSense/librealsense/issues/4034

ls880, 2021-04-15:
This is great, these comments have helped me find the error thank you! One last question; is there any way to find the number of depth frames, other than using the realsense viewer?

MartyG-RealSense, 2021-04-15:
Great news that you were successful with **get_real_time** in the MATLAB wrapper!

If you are using a bag file, you could check the depth topics with the SDK's **rosbag inspector** tool.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rosbag-inspector

closed 2021-04-27 by MartyG-RealSense