realsenseai/librealsense #8705: GL CallBack GL Error after running realsense-viewer

gapapadopoulos, 2021-03-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |  Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.8.0-48                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.43.0                          | 
| Language                            |  shell                         | 

### Issue Description
1. I went through this installation guide mentioned here https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md and after running realsense-viewer in terminal I get this error: "GL CALLBACK: ** GL ERROR ** type = 0x824c, severity = 0x9146, message = GL_INVALID_VALUE error generated. Width and height must not be negative".
2. My terminal looks like this below
[image]
3. My Realsense Viewer looks blank like this below
[image]
I am stuck and have no idea what to do if anyone can help that would be great!

MartyG-RealSense, 2021-03-31:
Hi @gapapadopoulos   I note fron the information in the header of this case that you are using kernel **5.8.0-48**.  The RealSense SDK currently officially supports kernel versions up to **5.4**.  Using newer kernel versions than 5.4 may therefore have unpredictable results.

If you are unable to use kernel 5.4, would it be possible please, instead of using the distribution packages, to try building the SDK from source code with CMake using the **RSUSB backend** installation method?  This method - which uses the CMake build fag **DFORCE_RSUSB_BACKEND=true** - requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.  Instructions for doing so are below.

*********************

1.  Visit the SDK **Releases** page and scroll down to the **Assets** file list at the bottom of the listing for each SDK version.  If there is only the word Assets there with an arrow beside it, left-click on the arrow to expand open the file list.

Download a zipped folder of the SDK's source code by left-clicking the link named **Source code**.

[image]

Once the file has downloaded, extract the contents of the zip file.

2.  Go to the **Librealsense** root directory of the folder extracted from the zip file and create a **Build** folder, and then change to that created folder using the command below:

**mkdir build && cd build**

3.  Whilst in the Build folder, run the CMake command below to install the SDK.

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

4.  Finally, run this CMake command:

**make && sudo make install**

gapapadopoulos, 2021-03-31:
Hello i downloaded the zip folder and followed the instructions and got a similar error shown below.
[image]

MartyG-RealSense, 2021-04-01:
Thank you very much for attempting an RSUSB installation.

Next, could you try updating the graphics drivers on your computer please to help eliminate the possibility that there is a problem with rendering the OpenGL graphics that the RealSense Viewer is dependent upon?

gapapadopoulos, 2021-04-01:
I have NVIDIA Driver version: 460.32.03 and my graphics card is GeForce GTX 1660 Ti. I purged Nvidia and redownloaded the driver. I still get the same error.

MartyG-RealSense, 2021-04-02:
As there is only one other recorded occurrence of this GL error with librealsense, let's go a step back in the log to the **Failed to open USB interface** and the **Failed to set power state** errors before it and see whether tackling those can improve the situation.

Some RealSense users have found that the advice in the link below has helped resolve their problem.

https://github.com/IntelRealSense/realsense-ros/issues/1408#issuecomment-698128999

closed 2021-04-14 by MartyG-RealSense