realsenseai/librealsense #8710: L515. NiR, DEPTH and RGB data are not synchronized with each other (frame numbers of NIR, DEPTH and RGB are different).

anhTuan0712, 2021-03-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.00 To 01.05.04.01 | 
| Operating System & Version |   Win (8.1/10)  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | Pc |
| SDK Version                            |    }                          | 
| Language                            |  C/C#                         | 
| Segment			|                           | 

### Issue Description
1. Config 
[image]
2. Get frame and print the number of frames of each type.
[image]
3.  Output: Frame number of them are not the same => NiR, DEPTH and RGB data are not synchronized with each othe.
[image]
[image]

### Question 
Why is there such difference? Please tell me how to solve this problem?

anhTuan0712, 2021-04-01:
@RealSenseSupport  please help me with this problem! Thank you!

RealSenseSupport, 2021-04-02:
Hi @anhTuan0712 

Just a couple of observations.  I notice that you're able to set the camera into the same resolution for all streams.  Our supported Depth, IR, Confidence resolutions are 320x240, 640x480, 1024x768.  The color resolutions that are supported are 960x540, 1280x720, 1920x1080

When I configure my camera using RealSense Viewer, the Depth, IR and Confidence streams are aligned.  When pausing the stream, you can see that the streams are aligned on frame number etc.  Color is a separate stream and I wouldn't expect the color stream frame count to be identical to that of the Depth, IR, Confidence streams.

I'm using the 2.43 RealSense Viewer and L515 camera FW 1.5.5 which is the latest FW.

anhTuan0712, 2021-04-05:
> Hi @anhTuan0712
>
> Just a couple of observations. I notice that you're able to set the camera into the same resolution for all streams. Our supported Depth, IR, Confidence resolutions are 320x240, [...]

I'm glad I received your answer!
I have the following 2 issues that need you to clarify:
1) Do you have any mistakes when you say "The color resolutions that are supported are 960x540, 1280x720, 1920x1080",when I check that color resolution 640x480 still supported on SDK 2.43 and FW 1.5.5?
[image]

2)  Yes I understood this issue " Color is a separate stream and I wouldn't expect the color stream frame count to be identical to that of the Depth, IR, Confidence streams." but why Depth and Ir are not aligned (asynchronous).
2.1) when I use the SDK in my application the number of the IR frame and the Depth frame number is sometimes not aligned (asynchronous).
2.2) Here is my code (I check if not aligned then stop program).
[image]

2.3) Here is the error output.
[image]
[image]

2.4 )The error will be more easily encountered if the computer CPU is high.

RealSenseSupport, 2021-04-14:
Hi @anhTuan0712 

Do you have any mistakes when you say "The color resolutions that are supported are 960x540, 1280x720, 1920x1080",when I check that color resolution 640x480 still supported on SDK 2.43 and FW 1.5.5?

As stated our official supported RGB resolutions are 960x540, 1280x720, 1920x1080.  If your camera is showing more than this, it is an anomaly.  Future FW releases may remove these unsupported resolutions, I just want to make sure you're aware of that.  

Regarding synced frames, I cannot see anything specific in your code that could cause this but I do not think any of our samples that keeps records of frames similar to that of the RealSense Viewer is having issues with missed frames or unsynced frames.

anhTuan0712, 2021-04-15:
Hi @RealSenseSupport .
1) This problem "As stated our official supported RGB resolutions are 960x540, 1280x720, 1920x1080. If your camera is showing more than this, it is an anomaly. Future FW releases may remove these unsupported resolutions, I just want to make sure you're aware of that" I understand, thank you. 
2) For this issue "Regarding synced frames, I cannot see anything specific in your code that could cause this but I do not think any of our samples that keeps records of frames similar to that of the RealSense Viewer is having issues with missed frames or unsynced frames.", it only shows up at a few moments and then it can be restored on its own, so it's unlikely that you can see the error in the Realsense Viewer (recommended above).
[image]
=> Please check like my code above.

anhTuan0712, 2021-04-22:
Hi @RealSenseSupport. 
On the "Intel realsense viewer" app, it's hard to see this error but luckily I've seen it.
[image]

anhTuan0712, 2021-05-10:
Hi @RealSenseSupport , Hi @MartyG-RealSense !
I am looking forward to an answer from you.

RealSenseSupport, 2021-06-09:
Hi @anhTuan0712 

We are still not able to reproduce this at all.  We also went ahead and updated the Viewer to 2.47 which is the latest to make sure we have the latest updates to the SDK package and updated the L515 FW.  Did notice you using the 2.43 release.  I'm wondering if it's a system level or USB level bandwidth issue that could be causing the frames to arrive delayed.

MartyG-RealSense, 2021-07-02:
Hi @anhTuan0712  Do you require further assistance with this case, please?

closed 2021-07-08 by MartyG-RealSense