realsenseai/librealsense #8711: Color Stream: No Frames Received

MichaelD92, 2021-03-31:
[image]
| | |
|---|---|
|**librealsense**|2.42.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|102422075653|
|**Firmware Version**|05.12.11.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|043323052498|
|**Firmware Update Id**|043323052498|

I have the problem that the RGB camera does not receive frames. When I display the frame metadata, the error "Per-frame metadata is not enabled at the OS level! Please follow the installation guide for the details" is displayed and the logged issues can be seen in the picture.

[image]
[image]

MartyG-RealSense, 2021-03-31:
Hi @MichaelD92   Would it be possible to update your RealSense SDK version to the current **2.43.0** version using the automated Windows installer file from the **Assets** file list at the link below to see whether the problem still occurs, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.43.0

[image]

In regard to the metadata problem: you should still be able to view the RGB stream if support for **Frame Metadata** has not been enabled.  The message **Per-frame metadata is not enabled at the OS level** should disappear if you enable metadata in the RealSense Viewer.

If metadata is not enabled, you may see the message box below when you launch the Viewer.  Clicking the **Enable** button on that message box adds metadata support.

[image]

If the message cannot be made to appear, it may be triggered by unplugging the camera from the USB port and plugging it into a different USB port if you have an additional one available.

****************************

May I also ask whether you have enabled the **Motion Module** at any time during a Viewer session when Depth and RGB have also been enabled?  Some RealSense users experience **No Frames Received** on the RGB stream if they have enabled Depth, RGB and Motion Module during the same Viewer session.  It does not matter which order the three stream types are activated in, because enabling the Motion Module is always the trigger for the issue.  

Once the No Frames Received problem with the RGB stream has been first triggered, it may persist even after the Viewer has been closed and re-launched, even if the Motion Module is not enabled in the subsequent sessions.  It can be corrected by a reboot of the computer.

MichaelD92, 2021-04-01:
I have now updated the firmware. Unfortunately, it still does not work. Somehow the message box does not appear even after several tries. I have plugged the camera into several USB ports and started the computer several times. I have also not activated the motion module.

MichaelD92, 2021-04-01:
I have just tried the camera on another PC. The box appeared there and I was able to click enable. However, the RGB stream does not work there either.

MartyG-RealSense, 2021-04-01:
Please check that the RGB camera is able to function by using the **Camera** app in Windows, as described in the instructions below.

STEP ONE
Please type the word **camera** into the text box at the bottom of your Windows screen to find the Windows Camera app.  This is an app designed for ordinary cameras.

[image]

**STEP TWO**
Run the Camera app with your RealSense camera plugged in.  

If the display is empty then left-click on the **Change Camera** option to switch to the RealSense camera.

If you get a pop-up warning saying that the camera could not be started but you can see the RGB video image behind the box, just ignore it and click **Close** to continue.

Do you see a normal video image like the one below when following this procedure?

[image]

MartyG-RealSense, 2021-04-07:
Hi @MichaelD92  Do you require further assistance with this case, please?  Thanks!

MichaelD92, 2021-04-07:
Currently I still need help. When I start the camera app, the error message appears on the picture. I did some Windows updates yesterday. Unfortunately, there is still no picture displayed.
[image]

MartyG-RealSense, 2021-04-07:
Thank you very much for the update.  Can you confirm please that you can receive no picture in the Camera app even if you toggle the **Change Camera** option repeatedly by left-clicking on it?

MichaelD92, 2021-04-08:
The camera does not show a picture and the error message appears even when I click on Change camera.

I have reinstalled the SDK and made sure to perform each step as administrator. Now this error message appears in Realsense Viewer
[image]

MartyG-RealSense, 2021-04-08:
I would expect to see the **CreateDeviceSource** error on Windows if permission for the camera to be accessed by applications was set to Off in the Windows privacy settings.  Instructions in the link belkow describe how to check this.  The access should be set to On.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

MichaelD92, 2021-04-09:
In the privacy setting for the camera is everything on "ON". Today, I uninstalled the SDK again and reinstalled everything as administrator. I could not reproduce the error from yesterday. The message box to enable the Frame-Metadata was also displayed and I was able to activate it. Unfortunately, this did not solve my problem as you can see in the pictures. Playback in the camera app is still not working. 
[image]
[image]
Is it possible that this is not a software problem but a hardware problem?

MartyG-RealSense, 2021-04-09:
I note that in the image of the depth stream above, you have RGB enabled too.  Have you enabled the Motion Module too at any point?  This can be a trigger in the Viewer for the RGB stream becoming No Frames Received and may not be corrected until the computer is rebooted (closing the Viewer and re-launching it does not resolve the problem).

If you have not enabled the Motion Module then you could try the standalone version of the Viewer - **Intel.Realsense.Viewer.exe** - that can be downloaded from the same **Assets** list that the SDK installer file is downloaded from.  This version of the Viewer does not contain computer processor minimum requirements that the Viewer bundled in the SDK has.

[image]

Once **Intel.RealSense.Viewer.exe** is downloaded to the **Downloads** folder of your computer after left-clicking on the link in the Assets list then it can be launched from that folder with a double-leftclick (do not use the Viewer shortcut icon on the Windows desktop).

MichaelD92, 2021-04-12:
When I got the camera, I already switched on the motion module once. But since then some time has passed and in the meantime I have uninstalled and reinstalled the SDK several times and also restarted the computer several times. I have now also downloaded the stand alone version of the viewer and uninstalled the SDK. Unfortunately, my problem has not been solved.

MartyG-RealSense, 2021-04-12:
As the camera is identified by the Viewer as D435i, this suggests that the camera firmware is correctly detecting the presence of the RGB sensor hardware, otherwise the firmware would likely identify the camera model as **D430i** (i,e no RGB).  

I note that you have uninstalled and reinstalled several times.  Does that mean that you have already tried the recent **2.44.0** SDK please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.44.0

MichaelD92, 2021-04-13:
I have now installed SDK 2.44. As you can see in the picture, I first started the Realsense Viewer without connecting the camera. The logged errors "types.h:315" appeared there.

[image]

Then I connected the camera. Then two warnings "DS5 group_devices is empty" appeared.

[image]

MartyG-RealSense, 2021-04-13:
Does your computer have an **AMD** processor?  The 400 Series cameras should work with any Intel or Arm processor.  There are so such guarantees for AMD-based computers though, and I experience more Viewer log errors on my AMD processor laptop than I do with my Intel processor laptop.

MichaelD92, 2021-04-14:
I use an Intel I7-9700 and an Nvidia Quadro P1000 as a graphics card. The PC also has 16 GB RAM.

MartyG-RealSense, 2021-04-14:
Thanks very much for the information.  Do the problems persist if you reset the camera hardware by going to the **More** option of the Viewer and selecting the **Hardware Reset** option from its menu?

[image]

MichaelD92, 2021-04-15:
Yes, the errors persist after a hardware reset. You can see the logged messages in the pictures.
[image]
[image]
[image]

MichaelD92, 2021-04-16:
I have now uninstalled the camera drivers and followed your instructions. I can trigger the message box to appear using this method. However, I still do not get any pictures from the RGB camera.
[image]
[image]
[image]

MartyG-RealSense, 2021-04-16:
In your **Cameras** section of the Device Manager, could you right-click on **Xi Imager** and select the menu option **Disable device** to test whether ths device is preventing the RealSense RGB from operating?  Or unplug it if it is a USB device.  

If disabling it does not make a difference then you can re-enable it by right-clicking on Xi Imager again and selecting **Enable device**.

MichaelD92, 2021-04-19:
By deactivating the Xi Imager, the logged errors have disappeared, but there is still no RGB image available. I noticed that when I unplug the Xi Imager and then plug it back in, the Viewer displays the message "Found Realsense devices". The Xi Imager is a thermal imaging camera from Micro Epsilon.
[image]
[image]

MartyG-RealSense, 2021-04-19:
What does the log show if you launch the Viewer with no cameras attached and then plug in the Xi Imager?

MichaelD92, 2021-04-20:
This is what happens when I start the Viewer with all cameras unplugged and then plug in the Xi Imager after startup.
[image]
[image]
[image]

MartyG-RealSense, 2021-04-20:
I tested the Viewer with an old RealSense F200 camera that is not supported by RealSense SDK 2.0.  Like your Xi Imager, the F200 also generated a **Found 1 RealSense devices** message in the log, even though the Viewer cannot generate options in the side-panel for that camera model.  So if this log message is generated even for unsupported cameras, it may epxlain why your Xi Imager causes this message when inserted with no other cameras attached.

So to confirm: even when the Xi Imager camera is unplugged, you cannot get RGB when the D435i is plugged in?

MichaelD92, 2021-04-20:
If the Xi Imager is not plugged in, but the Realsense is, I do not get an RGB image.
[image]

MartyG-RealSense, 2021-04-20:
I went through the case from the beginning and do not see a mention of trying a different resolution.  Can the camera stream if RGB is set to resolution **640x480** ?

MichaelD92, 2021-04-20:
No, I have now tried different settings with the resolution and the frame rate and no stream comes up with any of them.

MartyG-RealSense, 2021-04-20:
I do not think that we have tried a reset of the **Viewer** settings yet (which is different from a reset of the camera).  A reset of the Viewer configuration can be performed with the instructions in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360052013914/comments/1500000416522

MichaelD92, 2021-04-21:
I have reset the Viewer settings, unfortunately there is still no RGB stream.

MartyG-RealSense, 2021-04-21:
A camera can be reset to factory defaults using a tool called **CustomRW** that is packaged with the RealSense Dynamic Calibration tool.   The Windows version of the tool can be downloaded from the link below and installed.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

After installation is complete, you can input the command below into the Windows Command Prompt to launch CustomRW in **Gold Reset** mode:

**Intel.Realsense.CustomRW.exe -g**

MichaelD92, 2021-04-21:
I have reset the camera to factory settings, but the RGB stream is still not available.
[image]

MartyG-RealSense, 2021-04-21:
Has your D435i's RGB ever functioned since you purchased the camera, or did it work for a time and then just stop working one day?

MichaelD92, 2021-04-22:
The RGB camera worked at the beginning and at some point it no longer delivered a stream.

MichaelD92, 2021-04-23:
The other computer I tried the RGB camera on is the same model, but it doesn't have an Nvidia graphics card. The RGB camera does not work on the other computer either.

MartyG-RealSense, 2021-04-23:
Let's test the RGB with an example program to confirm if RGB is inaccessible in the examples too.

On the Windows desktop, right-click on the launch shortcut for the RealSense Viewer and select the **Open file location** menu option.  This will open the SDK **Tools** folder where the SDK's pre-built launchable example programs are stored.

Locate a file called **rs-capture.exe** and double-leftclick on it to launch the program.

[image]

With a D435i camera model, the **rs-capture** program generates a multi-panel view of the depth, RGB and IMU streams when the camera is working correctly, like this:

[image]

MichaelD92, 2021-04-26:
When trying to run the programme, the error "No feedback" appeared. The camera is recognised in the Viewer.
[image]

MartyG-RealSense, 2021-04-26:
Thank you very much.  At this point I will seek advice from Intel about your case.

MartyG-RealSense, 2021-04-26:
Could you send me an email please at the address below so that we have your email address.  I can then arrange the next step of the support process for your camera.  

martyx.grover@intel.com

MartyG-RealSense, 2021-05-03:
Hi @MichaelD92  Could you send me your email address to my email above please so that we can assist you.  Thanks!

MartyG-RealSense, 2021-05-03:
Thanks very much for sending your email address @MichaelD92 - I have passed it to the appropriate RealSense team member.

MartyG-RealSense, 2021-05-25:
Case closed due to resolution reached with customer.

closed 2021-05-25 by MartyG-RealSense

calmelo, 2021-08-01:
Have you solved the problem? I meet the problem too.

MartyG-RealSense, 2021-08-01:
Hi @calmelo  Please start a new case for your issue by visiting the front page of this support forum at the link below and clicking the green **New Issue** button.  Include a link to this case here in your message so that we can refer to it.  Thanks!

https://github.com/IntelRealSense/librealsense/issues