realsenseai/librealsense #8716: Unrectified Color Images D435 YUY2 to RGB

jrder, 2021-03-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  (n/a)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.41.02657                          | 
| Language                            |  MATLAB                         | 
| Segment			| others                         | 

### Issue Description
I'm trying to convert the unrectified color images from a D435, from the native YUY2 format to RGB format, for use in MATLAB for custom calibration. So far, I've de-interleaved the YUY2 into YUV channels. I'm unsure what standard to use to convert the YUV to RGB, specifically I'm not sure if or what gamma correction values to use.

I'm aware of the OpenCV cvtColor function, but I'm unfamiliar with git, so I'm unable to read the source code for reference.

Currently, I'm looking at the ITU.BT-709 HDTV standard. I've read a few sources with conflicting results. Also, the HDTV standard documentation is hard for me to interpret.

I'm hoping for some guidance on this issue, if not a complete walkthrough, since a simpler process might have been taken to convert the unrectified infrared images from Y12I to Y16.

The following are some sources I've been reading:

https://www.itu.int/rec/R-REC-BT.709-6-201506-I/en
https://www.intel.ca/content/www/ca/en/support/articles/000026725/emerging-technologies/intel-realsense-technology.html
http://www.martinreddy.net/gfx/faqs/colorconv.faq
https://stackoverflow.com/questions/7041172/pils-colour-space-conversion-ycbcr-rgb?rq=1

Thanks in advance.

MartyG-RealSense, 2021-04-01:
Hi @jrder  In the couple of cases I have experienced where raw unrectified RGB frames were being accessed from the camera hardware, **RAW16** was being used as the source format.

https://github.com/IntelRealSense/librealsense/issues/7275

https://github.com/IntelRealSense/librealsense/issues/8025

One of those cases involved a RealSense user who, like yourself, was trying to access the frames in MATLAB.  They found that they had problems if they saved the RAW16 frames to a bag file and then read them into MATLAB, but it worked fine if they used MATLAB to read the RAW16 frames from the camera directly.

jrder, 2021-04-01:
Hi @MartyG-RealSense I'm surprised that they were using a RAW16 format. The D435 datasheet and custom calibration whitepaper don't identify this as a supported format. Also, I'm reading the images from the device via the SDK and MATLAB, without the use of a bag file.

I eventually found a color space conversion that worked, but the colors are washed out. Since I'm using the images for calibration, to identify the corners of a checkerboard pattern, I might just use the Y channel from the images. Alternatively, I might just try to read the OpenCV cvtColor again.

Notably, I found it somewhat validating that color space conversion is not too simple as discussed in another issue tagged in 

#7275
opencv/opencv#13066

MartyG-RealSense, 2021-04-01:
Thanks very much for the update.   Yes, getting a good color match from a conversion can be tricky as https://github.com/IntelRealSense/librealsense/issues/7275 highlights.

Please do let me know which solution you decide on or whether you require further assistance.  Good luck!

jrder, 2021-04-05:
Thanks again for looking at this. My end goal was to perform custom calibration on my D435 using the YUY2 unrectified color images. I was able to get calibration parameters into a XML file. But I was unable to write to the device using CustomRW because I received 'error 2002 : left intrinsic are unreasonable'. I ended up just using the Dynamic Calibrator.

Below is my code for YUY2 to RGB8 format in MATLAB 2020b, which includes color conversion. Hopefully, this helps someone else out in the future.

```
function [RGB] = yuy2rgb(yuy2)
%% convert YUY2 to RGB
% written by jrder in MATLAB 2020b
% inputs: packaged and interleaved YUV 422 (YUY2 format)
% outputs: RGB8 format 
arguments
  yuy2 uint32
end

%% initialization
  % YUV channels
  Y = zeros(2*length(yuy2),1); Y = uint8(Y);
  U = Y; V = Y;
  % RGB channels
  RGB = zeros(2*length(yuy2),3); 
  
%% YUY2 to YUV444
  % vectorized typecast macropixels into bytes
  bval = typecast(yuy2,'uint8');
  % sort bytes into YUV channels
  Y(1:2:end) = bval(1:4:end);
  Y(2:2:end) = bval(3:4:end);
  U(1:2:end) = bval(2:4:end);
  U(2:2:end) = bval(2:4:end);
  V(1:2:end) = bval(4:4:end);
  V(2:2:end) = bval(4:4:end);
    
%% YUV444 to RGB
  % clamp YUV channels
  Y = Y-16;
  U = U-128;
  V = V-128;
  % conversion to RGB  
  RGB(:,1) = 1.164*Y + 1.596*V;           % R  
  RGB(:,2) = 1.164*Y - 0.392*U - 0.813*V; % G  
  RGB(:,3) = 1.165*Y + 2.017*U;           % B
  % conversion to uint8
  RGB = uint8(RGB);
end
```

closed 2021-04-05 by jrder

MartyG-RealSense, 2021-04-05:
Thanks so much @jrder for sharing your results in MATLAB 2020b with the RealSense community!