realsenseai/librealsense #8717: IR image on D455 is much darker then on D435i

dsvua, 2021-04-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i and D455 | 
| Firmware Version                   | 05.12.11.00 for both cameras | 
| Operating System & Version |  Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  5.8.0-48                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  Robot                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
There are few complains:
- Infrared image on D455 is generally much darker then on D435i. When I manually adjust `Advanced Controls -> AE Control` to the same value as on D435i, D455 depth quality increases. IR and Depth images:
D435i
[image]
[image]
D455
[image]
[image]
D455 IR image after manual AE Control adjustment
[image]
- another complain: table top is nicely visible on depth image on D435i and basically invisible on D455. For robot it is important to see floor as well 

Is it suppose to be like that, or it is just my D455 camera misbehave? Both cameras are connected to the same computer next to each other.

MartyG-RealSense, 2021-04-01:
Hi @dsvua   I was able to replicate your infrared results with my own D435i and D455 side by side.  Disabling auto-exposure or instead changing the **Mean Intensity Set Point** value of AE Control on the D455 (as you suggested) made the infrared images of the two cameras more equal.

I found in my own tests that once both cameras were set to the same Mean Intensity Set Point of **1536** then their infrared images of a table-top were almost identical.

dsvua, 2021-04-01:
Yeah, but no depth for table top on D455. Like on attached depth images, you can see that on table top is covered in blue on depth image for D435i, when D455 have very sparse little depth islands.

MartyG-RealSense, 2021-04-01:
Please try maximizing the **Laser Power** option to '360' on the D455 controls to see whether it makes a positive difference.  the sparseness of the depth image is tied to the value of Laser Power.  A lower value increases sparseness (more holes / gaps) and a higher value reduces sparseness.

[image]

dsvua, 2021-04-01:
Almost the same. Plus I tried changing laser power when posted pictures - no visual effect on Depth image on D455

MartyG-RealSense, 2021-04-01:
I can see how the reduced visibility of the IR dot pattern on the desk on the D455 image could impact the ability of the camera to read depth detail from it.  Surfaces such as desks, walls and doors typically have low texture detail or no texture, so having a dot pattern projected onto such surfaces because the camera can use the dots as a texture source to analyze for depth detail.  So if the dot pattern on a low-texture surface has faint visibility or no visibility then that will have an impact on the depth image due to the reduction in analyzable depth detail.

Black or dark grey surfaces are also generally difficult for the camera to analyze because the dark color absorbs lights.  The darker the color, the more the light is absorbed by the surface.  Hence why a white wall has much brighter dots.  If a strong light source is cast on to a black surface (e.g a leather office chair) then it can help to bring out texture detail and make the illuminated areas more blue instead of being empty of depth detail.  The principle is illustrated in thhe link below.

https://github.com/IntelRealSense/librealsense/issues/6757

dsvua, 2021-04-01:
I understand that, but why D435i has no problem, when D455 cannot produce depth on table top? What is the difference between cameras?

MartyG-RealSense, 2021-04-01:
D455 has a minimum depth sensing distance of 0.4 meters, whilst on D435i it is just 0.1 meters (meaning that it can get closer to surfaces before the depth image starts breaking up).  So at close-range, the D455 may have less detail on the depth image in the foreground.  This difference gives the D455 a greater **maximum** depth sensing range.

You can reduce the minimum depth distance of the camera, whilst reducing its maximum depth distance as a trade-off, by increasing the value of the **Disparity Shift** option.  Setting the disparity shift to '50' should provide a good balance between gain in foreground and loss of background.

MartyG-RealSense, 2021-04-08:
Hi @dsvua  Do you require further assistance with this case, please?  Thanks!

dsvua, 2021-04-08:
Can you confirm that my understanding is right:

1.  D455 has generally darker IR images then D435i and it is normal.
2. Worst D455 performance (in comparison to D435i) for floor distance detection is due to wider IR cameras distance and related to it, worst, close to camera, depth detection.

Technically, this thread could be closed.

MartyG-RealSense, 2021-04-08:
I do not have any information to add about the infrared beyond what has already been mentioned.

In regard to the depth and the reduced detail on the desk near to the camera, this is symptomatic of the results that might be expected from the D455's larger minimum depth sensing distance (0.4 meters on D455 instead of 0.1 m on D435i).  Basically, a surface / object has to be further away from the camera lens on D455 for it to be rendered on the depth image.

As mentioned in https://github.com/IntelRealSense/librealsense/issues/8717#issuecomment-812131672 above, the camera's minimum depth sensing distance (MinZ) can be reduced by increasing the **Disparity Shift** option's value, enabling more of the near-range depth detail to be rendered on the depth image.

closed 2021-04-15 by dsvua