realsenseai/librealsense #8718: Is it possible to sync SR300 with other cameras

yuhongju1111, 2021-04-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Operating System & Version |   {Win 10 / Linux (Ubuntu 16)  | 
| Platform                                 | PC  |
                    | 

### Issue Description
I want to capture images simultaneously with SR300 and another RGB camera, is this possible? Does SR300 support external trigger as I found for D415? Or is it possible to set the trigger time for SR300 through the UVC protocol?

Thanks!

MartyG-RealSense, 2021-04-01:
Hi @yuhongju1111  The SR300 does not have any hardware pins for trigger / hardware sync like modern RealSense cameras do.

There was an SR300 project from 2017 that took the approach to capture triggering of listening on a **UDP socket** for a capture message from the application.

community.intel.com/t5/Items-with-no-label/Many-SR300-One-IR-light-While-Looking-at-Same-Object-i-e-Use-Two/m-p/462242#M3971

If you are recording footage from the RealSense and non-RealSense camera simultaneously and then syncing up the recorded footage afterwards instead of doing the syncing in real-time then using **frame metadata** to do so may be appropriate.

https://github.com/IntelRealSense/librealsense/issues/2186

As another possible alternative ... in January 2018, Intel demonstrated a system with D435 cameras where the start and end of capture were triggered by a commercial grade flash.  The spike in camera data caused by the flash acted as the trigger.   If your non-RealSense RGB camera has a flash then that sounds like a means by which a quick capture could be taken by the SR300, assuming that its camera data responds to a light-flash impulse by registering a spike in the same way a 400 Series stereo depth camera would (as SR300 uses a different depth sensing technology called **coded light**).

https://www.intelrealsense.com/intel-realsense-volumetric-capture/

MartyG-RealSense, 2021-04-08:
Hi @yuhongju1111  Do you require further assistance with this case, please?  Thanks!

yuhongju1111, 2021-04-13:
Thanks a lot @MartyG-RealSense. I just encountered another problem . i'm trying to get the hardware clock in global time domain.  but it seems that my SR300 doesn't support global time domain.  

sensor. supports(RS2_OPTION_GLOBAL_TIME_ENABLED
) returns false

is this just not supported in SR300 or is it just i’m not using the sdk correctly?

Thanks

MartyG-RealSense, 2021-04-13:
You are correct, Global Time Enabled is not provided as a supported option for the SR300.

https://github.com/IntelRealSense/librealsense/issues/4822

[image]

closed 2021-04-26 by MartyG-RealSense