realsenseai/librealsense #8719: How to record specific topics L515 camera into bag file by using librealsense SDK without ROS?

SatishAkula91, 2021-04-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseSupport, 2021-04-06:
@SatishAkula91 The default only includes depth streaming for rs-record-playback. Please add color streaming (e.g. color = color_map.process(frames.get_color_frame());) to the example below, then you can get both depth and color streaming.
https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp#L72

Also please modify the rendering and playback accordingly.
https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp#L197
https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp#L167

SatishAkula91, 2021-04-06:
@RealSenseSupport  Thanks for the support.

RealSenseSupport, 2021-04-09:
@SatishAkula91 Sorry. My mistake. So I understood you don't want to record motion data into a.bag, right? 
You can try to add two lines below after 
https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp#L94

cfg.enable_stream(rs2_stream::RS2_STREAM_COLOR, 640, 480, rs2_format::RS2_FORMAT_RGB8, 30);
cfg.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 640, 480, rs2_format::RS2_FORMAT_Z16, 30);

SatishAkula91, 2021-04-12:
@RealSenseSupport  Thanks for the support.

It's working now.

RealSenseSupport, 2021-04-12:
@SatishAkula91 Thanks for the update! Glad to hear that it works. Close this ticket accordingly.

closed 2021-04-12 by RealSenseSupport