realsenseai/librealsense #8720: Y-axis orientation in Intel Realsense Viewer

anhTuan0712, 2021-04-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515 | 
| Firmware Version                   |  01.05.03.00 | 
| Operating System & Version |   Win (8.1/10)  | 
| Kernel Version (Linux Only)    |                                    | 
| Platform                                 | PC|
| SDK Version                           |   2.36.0                        | 
| Language                               |                            | 
| Segment			        |                        | 

### Issue Description
1) Looking at the following image I see the Y-axis is facing downwards.
[image]
2) Issue https://github.com/IntelRealSense/librealsense/issues/3589, @RealSenseCustomerSupport said: "This is expected as starting from SDK 2.19.0, the points are converted from right hand to left hand coordinate system when exporting vertices to ply."
[image]
3) But as far as I know with right hand or left hand,  the y axis is always pointing up. (https://www.evl.uic.edu/ralph/508S98/coordinates.html)
### Question 
1) So what standard is the axis system in Realsense Viewer?
2) In the API I also had a similar question in issue https://github.com/IntelRealSense/librealsense/issues/8656, I still expect an answer on this matter.
Hi @RealSenseSupport , @RealSenseCustomerSupport !
Please help me with this problem! Thank you!
<Describe your issue / question / feature request / etc..>

RealSenseSupport, 2021-04-02:
Hi @anhTuan0712 

From the RealSense Viewer standpoint, the positive Y axis is in the downward and the positive X axis is to the right.

I'd like to point you to the documentation that we have here:  https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md

Also how our T265 camera is oriented here:  https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md

Another set of documentation to review:  https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#point-coordinates

IMU calibration tool:  https://github.com/IntelRealSense/librealsense/tree/development/tools/rs-imu-calibration#rs-imu-calibration-tool
https://www.intelrealsense.com/wp-content/uploads/2020/07/IMU_Calibration_Tool_for_Intel_RealSense-Depth_Cameras_Whitepaper.pdf?_ga=2.52513256.1165655506.1617391504-1479303487.1578600965

I think there may be confusion so the documentation for the cameras should help here.

anhTuan0712, 2021-04-05:
> Hi @anhTuan0712
>
> From the RealSense Viewer standpoint, the positive Y axis is in the downward and the positive X axis is to the right.
>
> I'd like to point you to the documentation that we have here: [...]

Yes, thank you for the feedback!
Everything you said above I understand but please answer me this problem https://github.com/IntelRealSense/librealsense/issues/8656

RealSenseSupport, 2021-04-10:
Hi @anhTuan0712 

Please see comments here:

https://github.com/IntelRealSense/librealsense/issues/8656#issuecomment-817067034

Thanks

anhTuan0712, 2021-04-12:
> Hi @anhTuan0712
> 
> Please see comments here:
> 
> [#8656 (comment)](https://github.com/IntelRealSense/librealsense/issues/8656#issuecomment-817067034)
> 
> Thanks
Hi @RealSenseSupport !
Thank you you have feedback.

RealSenseSupport, 2021-04-12:
Closing as remaining issue being tracked on https://github.com/IntelRealSense/librealsense/issues/8656

closed 2021-04-12 by RealSenseSupport