realsenseai/librealsense #8721: D435 RuntimeError: No device connected

korin-lf, 2021-04-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                     D435
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |Ubuntu 20.04
| Kernel Version (Linux Only)    5.4                                      | 
| Platform                                 | PC/
| SDK Version                                    | 
| Language                        python                     | 
| Segment			                  | 

Real sense viewer Recognises both T265 and D435
On pycharm no connection problem for T265 but RuntimeError: No device connected  for D435

RealSenseSupport, 2021-04-06:
@korin-lf Did you get such issue with your own code or python examples? https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples 

And any issue out of pycharm?

korin-lf, 2021-04-06:
Yes I tried Python examples in Pycharm.
The T camera works.
And D435 gives runtime error

RealSenseSupport, 2021-04-06:
@korin-lf Could you please tell which example you tested when get the issue? Also please provide more information below.

SDK(librealsense) version:
Firmware version:

RealSenseSupport, 2021-04-20:
@korin-lf Could you please provide more information so that we can further support? Thanks!

RealSenseSupport, 2021-04-29:
@korin-lf Do you still have issue? Could you please update? Thanks!

RealSenseSupport, 2021-05-12:
@korin-lf Sorry that we don't hear from you for weeks. Will close the ticket as this point. Please feel free to create another one if you still have issues or questions. Thanks!

closed 2021-05-12 by RealSenseSupport