realsenseai/librealsense #8722: How to convert the camera coodinates of any object  to the world coordinates  by only one D435i?

yuanyuanhan327, 2021-04-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-04-01:
Hi @yuanyuanhan327  It sounds as though you need to **deproject** the coordinates by converting 2D pixel coordinates to 3D world coordinates.  A RealSense SDK instruction that can do this is **rs2_deproject_pixel_to_point**

So that I can recommend a suitable scripting reference to you, can you tell me please which programming language are you using please (C++, Python, C#, etc).  Thanks!

yuanyuanhan327, 2021-04-01:
Thanks. I am using python. I know how to deproject the coordinates by converting 2D pixel coordinates to 3D coordinates in camara coordinates system.  But the camera is always moving. I don't know how to get the extrinsic parameters for the D435i  camaera at the real time.

MartyG-RealSense, 2021-04-01:
The link below has a Python script provided by a RealSense team member for retrieving intrinsic and extrinsic values and printing the values.  Does this meet oyur needs please?

https://github.com/IntelRealSense/librealsense/issues/2204#issuecomment-414497056

[image]

closed 2021-04-14 by MartyG-RealSense