realsenseai/librealsense #8723: Not work L500 Depth Sensor

webant-ltd, 2021-04-01:
| | |
|---|---|
|**librealsense**|2.43.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0245197|
|**Firmware Version**|01.04.01.02|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003b52af7f7|
|**Firmware Update Id**|0003b52af7f7|

I'm using the USB cable that came with the L515.
Run `realsense-viewer`

After launching, I get an error
```
Invalid Value in rs2_open_multiple(sensor:0x7f6c78025d30,
profiles:0x7f6c782b2970, count:2):
hwmon command 0x2b( 4 80000000 0 0 ) failed (response -29= Value out of
range)
```

RGD Camera and Motion Module work stably.

webant-ltd, 2021-04-01:
This is output console
```
02:23:21 [Info]  - Starting streaming of Depth, Infrared...
02:23:21 [Info] ~/librealsense-2.44.0/src/sensor.cpp:1416 - Request: Y8 Infrared, 
Resolved to: Y8 Infrared, 
02:23:21 [Info]~/librealsense-2.44.0/src/sensor.cpp:1416 - Request: Z16 Depth, 
Resolved to: Y8 Infrared, Z16 Depth, 
02:23:21 [Error] ~/librealsense-2.44.0/src/types.h:315 - xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
02:23:21 [Error] ~/librealsense-2.44.0/src/l500/l500-depth.cpp:714 - Exception caught in l500_depth_sensor::open
02:23:21 [Error]  - a0 e3 15 10 0a 04 a5 20 2d 00 00 00 00 00 00 00 75 62 ca f7 
02:23:21 [Error]  - a0 e3 15 10 0a 04 a5 30 2d 00 00 00 00 00 00 00 db 63 ca f7 
02:23:21 [Error]  - a0 e3 15 10 0a 04 a5 40 2d 00 00 00 00 00 00 00 41 65 ca f7 
02:23:22 [Debug]  - a0 e1 1d 20 f4 00 09 52 03 03 81 00 00 00 00 00 18 4e cd 00 
02:23:23 [Debug]  - a0 e1 1d 20 f4 00 09 62 03 03 81 00 00 00 00 00 c2 06 ae 0c 
02:23:24 [Debug]  - a0 e1 1d 20 f4 00 09 72 03 03 81 00 00 00 00 00 6f 3d 8f 18 

```

webant-ltd, 2021-04-01:
```
root: /usr/local/bin/rs-hello-realsense 
RealSense error calling rs2_pipeline_start(pipe:0x55de394f2860):
    xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
```

```
root: /usr/local/bin/rs-hello-realsense
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense L515
    Serial number: f0245197
    Firmware version: 01.05.05.00

The connected device doesn't support depth streaming!
```

taletovic, 2021-04-06:
Hi @webant-ltd. I am having exactly the same problem - same errors are being shown in RealSense Viewer and when launching ROS Node. Funny thing is that it actually worked last week just normally. Can you confirm that as well or your sensor never worked as supposed? Thanks.

theotromp97, 2021-04-13:
@taletovic I have the exact same thing, my camera worked last week. I have been having success with it.
I might have messed something up during installation of realsenselib and now it does not work anymore.
I also have a problem regarding USB 3.0 port. When i unplug the USB C on the camera and rotate in 180 degrees it gets recognised as USB 2.1 instead of 3.0.

taletovic, 2021-04-13:
@theotromp97 I was using different presets (close range was interesting for me) as well as trying to save the settings in a .json file which could be later on loaded in the ROS driver. Maybe that could cause this problem, however I've no solution to it.

@RealSenseCustomerSupport any answer/comment on this? Please help.

EZIO7000, 2021-04-15:
I have this problem too, I followed the installation for the newest version and in the end when I try to turn on the L500 Death Sensor I have error:
"Backend in rs2_open_multiple(sensor:0x7fa8ac01bca0, profiles:0x7fa8ac41c490, count:2):
xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error". 
[image]

talhakhan13, 2021-04-27:
Did anyone figure out how a solution to this? It was working fine last week but now I'm getting the exact same errors as @webant-ltd when trying to turn on the depth sensor. The motion module and rgb camera are working fine though. I've even tested it out on a different computer but the problem persists.

RealSenseSupport, 2021-05-10:
Hi All

If you're having issues with the streaming of the depth in the manner outlined above or even having issues with the USBC connection orientation changing behavior with regards to detected as USB2 or USB3, please open a ticket here:  https://support.intelrealsense.com/hc/en-us/requests/new

closed 2021-06-09 by RealSenseSupport