realsenseai/librealsense #8725: pointcloud processing block Using too much memory and memory leak

anhTuan0712, 2021-04-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515 | 
| Firmware Version                   |  | 
| Operating System & Version |   Win (8.1/10)   | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 |   |
| SDK Version                            |  legacy / 2.36.0 and 2.42.0                     | 
| Language                            |  C#                          | 
| Segment			|                            | 

### Issue Description
1) when I use the processor blocks below it takes up a lot of memory, most times 2680MB.
[image]
[image]
2) when I am not using the above handle, my application is less than 60MB.
[image]

3) I referenced the issues: https://github.com/IntelRealSense/librealsense/issues/2860, https://github.com/IntelRealSense/librealsense/issues/5119, https://github.com/IntelRealSense/librealsense/issues/6734But the problem is still not resolved.
### Question 
Hi @RealSenseSupport , @MartyG-RealSense 
Let me know, what should I do to solve this problem?
<Describe your issue / question / feature request / etc..>

RealSenseSupport, 2021-04-08:
Hi @anhTuan0712 

Thanks for notifying us on this.  It seems as though this is related and the same as the other items that you've linked it to.  Some of those issues are still open.  Will monitor those as well.

anhTuan0712, 2021-04-12:
Hi @RealSenseSupport.
Thank you you have feedback.
Please notify me when this issue is resolved.

MartyG-RealSense, 2021-07-02:
Hi @anhTuan0712  Do you require further assistance with this case, please?  The case will be closed after 7 days from the time of writing this if we do not hear from you.  Thanks!

closed 2021-07-08 by MartyG-RealSense