realsenseai/librealsense #8727: Facemapping 2

kinectsources, 2021-04-03:
Will Facemapping 2 works with librealsense SDK ? As their repo .md updated few years ago there's nothing mentioned on new Realsense depth D4xx cameras. If so, will it work for android phones ?

About Facemapping 👇
https://github.com/GameTechDev/FaceMapping2

MartyG-RealSense, 2021-04-03:
Hi @kinectsources   The **Intel RealSense 3D Scan module** mentioned on the Facemapping2 GitHub was part of an old RealSense SDK for Windows that was published in 2014-2016.  It has no compatibility with the modern librealsense SDK as these two SDKs have totally different architectures.

I understand the kind of technology that you are referring to for applying a real face scan to a 3D mesh, which was used in a few Xbox Kinect games in 2015-2017 to varying degrees of success (the scan could be distorted by strong facial expressions or moving aroud too much during scanning).

[image]

A more consistently successful approach to Kinect face scan was one that constructed the face model from a **point cloud** type scan:

https://www.youtube.com/watch?v=o0g9RgbGnMI&t=26s

That point cloud approach can be replicated with modern 400 Series cameras in librealsense, converting the point cloud scan to a solid 3D model.

kinectsources, 2021-04-03:
I have found Unreal Engine blueprint for new Intel Realsense depth cameras. So, Can I implement in game to change player face ?

MartyG-RealSense, 2021-04-03:
The features supported by the librealsense SDK's Unreal wrapper are listed in the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unrealengine4#overview

Yes, you can generate and render a point cloud mesh with the wrapper.  And the documentation states that the point cloud uses dynamic mesh generation, which sounds useful for face morphing.  It is not obvious though how individual points could be scaled and repositioned to allow customization of the facial features.

MartyG-RealSense, 2021-04-11:
Hi @kinectsources  Do you require further assistance with this case, please?  Thanks!

kinectsources, 2021-04-11:
No thanks. You can close it

closed 2021-04-11 by kinectsources