realsenseai/librealsense #8728: Issue in app while building via dependency

bjain409, 2021-04-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

bjain409, 2021-04-03:
> * Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):
>
>   * Consider checking out SDK [...]

 @ev-mp
 "  https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/java_example"
 
By following these examples the application doesn't work.
Frames are not passed to android.
Using dependency is not working

RealSenseSupport, 2021-04-07:
@bjain409 What's your SDK(librealsense) version? And what kind of problem did you get when run the application? Did you follow https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/readme.md#project-configuration to modify accordingly?  I just tried the example and can run normally. Attached is the snapshot.
[image]

bjain409, 2021-04-07:
> @bjain409 What's your SDK(librealsense) version? And what kind of problem did you get when run the application? Did you follow [...]

@RealSenseSupport 
I have shared the repo address. Its the same code that is suggested
https://github.com/bjain409/libRealsenseViaDependency

bjain409, 2021-04-07:
> @RealSenseSupport
> I have shared the repo address. Its the same code that is suggested
> https://github.com/bjain409/libRealsenseViaDependency

@RealSenseSupport 
In the code that I shared, I am following the same thing. I copied the code of capture module Just to verify but it was not helpful

RealSenseSupport, 2021-04-08:
@bjain409 Please provide the issue snapshot and logcat logs.

bjain409, 2021-04-08:
> @bjain409 Please provide the issue snapshot and logcat logs.

@RealSenseSupport 
After building and running the app(the github link which I shared )
I don't get pop - up twice. I just get the pop up once
and It shows still shows plz attach inteldevice

bjain409, 2021-04-08:
> > @bjain409 Please provide the issue snapshot and logcat logs.
> 
> @RealSenseSupport
> After building and running the app(the github link which I shared )
> I don't get pop - up twice. I just get the pop up once
> and It shows still shows plz attach inteldevice

@RealSenseSupport if possible can you build the app on your side?
https://github.com/bjain409/libRealsenseViaDependency 

You can verify that I have followed the same instruction.

RealSenseSupport, 2021-04-08:
@bjain409 Got your issue now. What's your android version(8.1, 9 or 10)?  And can you run below applications normally on your android device? https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/2.44.0/camera-2.44.0.apk

bjain409, 2021-04-08:
> @bjain409 Got your issue now. What's your android version(8.1, 9 or 10)? And can you run below applications normally on your android device? https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/2.44.0/camera-2.44.0.apk

its 10.
I will check with given apk

bjain409, 2021-04-08:
> > @bjain409 Got your issue now. What's your android version(8.1, 9 or 10)? And can you run below applications normally on your android device? https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/2.44.0/camera-2.44.0.apk
> 
> its 10.
> I will check with given apk

I just installed the given apk.
It having the same issue.
I get the pop up only once for intel device 
And shows connect a realsense device.

bjain409, 2021-04-08:
> > > @bjain409 Got your issue now. What's your android version(8.1, 9 or 10)? And can you run below applications normally on your android device? https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/2.44.0/camera-2.44.0.apk
> > 
> > 
> > its 10.
> > I will check with given apk
> 
> @RealSenseSupport
> I just installed the given apk.
> It having the same issue.
> I get the pop up only once for intel device
> And shows connect a realsense device.

@RealSenseSupport 
If you help  with building
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/capture
via dependency then I will edit the code as per my requirements.

RealSenseSupport, 2021-04-08:
@bjain409 If you're getting the issue on Android 10, then I think it should be Android 10 issue which you can refer to https://issuetracker.google.com/issues/139087809 

Please try to change targetSdkVersion to 27(any below 28) and rebuild.  Thanks!

bjain409, 2021-04-08:
> @bjain409 If you're getting the issue on Android 10, then I think it should be Android 10 issue which you can refer to https://issuetracker.google.com/issues/139087809
> 
> Please try to change targetSdkVersion to 27(any below 28) and rebuild. Thanks!
@RealSenseSupport 
Then it should not work in this case as well right?
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/capture

RealSenseSupport, 2021-04-10:
@bjain409 For https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/capture you should also modify targetSDkVerson to 27(the line below) and rebuild. 

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/build.gradle#L8

bjain409, 2021-04-12:
Ok, this is working.
Thank you

closed 2021-04-12 by bjain409