realsenseai/librealsense #8730: RealSense Viewer not starting

zebrassimo, 2021-04-03:
| | |
|---|---|
|**librealsense**|2.43.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```
Depth quality tool is working, the RealSense viewer crashed.
[image]

MartyG-RealSense, 2021-04-03:
Hi @zebrassimo   This message indicates that the age of the processor in your computer makes it incompatible with the RealSense Viewer tool that is bundled in the RealSense SDK.  This Viewer seeks an Intel Core 6th Generation processor or newer.  Your E5-2620 processor is 2nd Generation.

You can download a standalone version of the Viewer from the **Assets** file list at the link below though that does not have this compatibility check.  

[image]

Please right-click on the **Intel.RealSense.Viewer.exe** link in the list and select the file download option from the menu (it may have a name such as **Save Link As** or **Save Target As**).

When the download has completed, go to the folder location that you downloaded the file to and double-leftclick on the file to launch the Viewer.  Do not use the launch shortcut on your Windows desktop, as this leads to the version with the compatibility check.

If you need a shortcut to the downloaded Viewer to place on your desktop, delete the original shortcut and make a new one by right-clicking on the file and selecting the **Create shortcut** menu option.

zebrassimo, 2021-04-03:
Thanks, this resolves the issue.

closed 2021-04-03 by zebrassimo

MartyG-RealSense, 2021-04-03:
You are welcome @zebrassimo - thanks for the update!