realsenseai/librealsense #8732: is the  high speed capture mode  implemented for the L515 series?

AndreV84, 2021-04-03:
is the  high speed capture mode  implemented for the L515 series?

RealSenseSupport, 2021-04-09:
Hi @AndreV84 

Can you explain more as to what you mean by high speed capture mode?

AndreV84, 2021-04-09:
@RealSenseSupport 
what is the highest fps rate?
what is the highest fps rate for infra sensor?
what is the highest fps for other sensors? for depth sensor?

AndreV84, 2021-04-09:
reference https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435

RealSenseSupport, 2021-04-21:
Hi @AndreV84 

For L515 the FPS for depth, infrared, confidence is fixed at 30fps
RGB is able to reach 60fps

This definitely differs from the D400 depth cameras

AndreV84, 2021-04-21:
but the throughput/ usb speed would allow a higher fps , no?

RealSenseSupport, 2021-05-13:
By design we fixed the FPS for the L515 camera product.

AndreV84, 2021-05-23:
@RealSense-Customer-Engineering 
can we use
modes with l515 from the list below?
```
, rs-server supports RGB at 640x480 (6, 15 or 30 FPS) or 424x240 (6, 15, 30 or 60 FPS).
```
can we use rs-server with L515?
How at all we could stream L515 over the network?

AndreV84, 2021-05-23:
@RealSenseCustomerSupport How to read in Linux from l515 using v4l2? is it possible?
is it possible to stream from l515, but also accessing it from local device?

RealSenseSupport, 2021-06-10:
@AndreV84 
Couple of items here, in the latest FW1.5.8.1/SDK2.47 we've added new RGB resolutions 640x480 and 640x360
Not sure I see a reason why L515 cannot stream over network.

AndreV84, 2021-06-10:
yes, we found l515 can stream from rs-server to network realsense viewer
However, is there some sort of rtsp steam that could be played e.g. with any generic rtsp player at the clients side or realsense-viewer is necessarily required at the client sode?

RealSenseSupport, 2021-06-21:
A RealSense SDK based app on the other end would be needed.

MartyG-RealSense, 2021-09-11:
Thanks very much @AndreV84 for the update - it's good to hear that a resolution was achieved!

AndreV84, 2023-08-09:
@MartyG-RealSense 
@RealSenseSupport 
Could you let me know if there were more recent firmware released for l515 than firmware version: 01.05.08.01, ?
Thanks

MartyG-RealSense, 2023-08-09:
Hi @AndreV84   No, there was no further L515 firmware released after 1.5.81.

AndreV84, 2023-09-05:
@MartyG-RealSense is there information about planned end-of-life  EOL  of d400 series or specific models of d4xx?

MartyG-RealSense, 2023-09-06:
There are no current plans for EOL of the 400 Series or specific models.  

Support for already EOL'd models such as L515 and SR300 is being removed from the ROS2 wrapper. The wrapper already now only supports ROS2 Humble and Iron.