realsenseai/librealsense #8734: OpenNI2 848x480 image is scrambled for both depth and RGB

noirflux, 2021-04-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435, D455 | 
| Operating System & Version |   Win 7, 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.42.0     | 
| Language                            |  OpenNI2           | 

### Issue Description
When I try to set the depth and color resolutions to 848x480 in a program using OpenNI2 (stopping and starting the streams with a full OpenNI2 restart), the resulting images are scrambled, like the row/column indexing or stride is wrong. 1280x720 works fine too, so it's not just an aspect ratio issue.

What I get at 640x480:
[image]

and 848x480:
[image]

This is with both a D435 on both Win7 and Win10, and a D455 on Win10,  with the drivers for each built from 2.42.0.

noirflux, 2021-04-04:
I thought maybe this was a result of the wrapper always reporting the camera as a Primesense, but I commented out the compiler flag for that and rebuilt the wrapper/driver, and same problem.  In all cases the OpenNI2 stream resolution is being reported correctly as 848x480, which is what is used to do the image copy in my program, so the scrambling looks like it is already in the buffer being given to my application.  I have tried to see where the image copy is done in the wrapper, but have not yet been able to trace everywhere the image resolution is being used (or not!) yet.

MartyG-RealSense, 2021-04-04:
Hi @noirflux  There is not much information on this OpenNI2 issue.  The only occurrence of it on the librealsense forums that I could find was the same text as the opening message, posted by @mediadog in the comment linked to below.

https://github.com/IntelRealSense/librealsense/issues/4138#issuecomment-809061773

Searching more widely outside of the librealsense forums though, I did find a discussion about problems with changing resolution in OpenNI2 when using both depth and color (single-stream configurations were not affected).  A commenter on that discussion found that they could resolve the problem if they started and stopped, **then** made the resolution change and started again.

https://stackoverflow.com/a/36273413

noirflux, 2021-04-05:
Yeah that was me, sometime I forget to check which account is logged in...  As I noted this is happening at 848x480 when I do a full fresh startup of OpenNI2.  I have gone over the code on my side looking at how the image is copied, and everything looks correct (and works with all other resolutions and other OpenNI2 supported devices).

I am very suspicious about the stride though, as in my code I directly compute it from the resolution and image type (bytes per pixel) while the wrapper internally appears to use a precomputed value for it.  If that stride was not the obvious value (edge padding or cropping for some reason) then that could throw my copy off.  And it's very odd this happens with both the depth and RGB images.

I'm still wading wading down through the code to see what could be happening, but it would be swell if some other OpenNI2 program could be used to verify or refute this.  I have yet to find if the resolution can be changed with the NiViewer program without rebuilding it.

MartyG-RealSense, 2021-04-06:
The section of the OpenNI2 documentation highlighted in the link below comments on FOV.

https://github.com/occipital/OpenNI2/blob/master/Include/OpenNI.h#L2377-L2384

noirflux, 2021-04-06:
Think you meant to put that in the FoV thread, but thanks.  This is why an accurate FoV is important though, as I am doing the depth data to world data in a computer shader for performance reasons.  If it is not correct, not only is the data not accurate, but it makes it impossible to combine pointclouds from multiple cameras.

MartyG-RealSense, 2021-04-12:
Hi @noirflux  Do you require further assistance with this case, please?  Thanks!

noirflux, 2021-04-12:
It is still a problem.  Is there anything being done to fix or at least verify it?  Any steps I need to take to help that process?  Can you point me to an OpenNI2 program that shows otherwise?

MartyG-RealSense, 2021-04-12:
As mentioned in my comment https://github.com/IntelRealSense/librealsense/issues/8739#issuecomment-818032933 on your other OpenNI2 case, the wrapper seems to be maintained in an "As Is" state where it works but receives few updates.  If you are able to develop additions for the wrapper yourself then you are welcome to contribute them for integration into the wrapper by filing a Pull Request.

https://github.com/IntelRealSense/librealsense/pulls

MartyG-RealSense, 2021-04-18:
Hi @noirflux  Bearing in mind the above comment about the infrequency of OpenNI2 wrapper updates, do you wish to continue this case?  Thanks!

closed 2021-04-24 by MartyG-RealSense