realsenseai/librealsense #8735: D435i FOV and Aspect Ratio

pranav-pang, 2021-04-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

So we're trying to estimate distance and height of certain colored objects (red) and we're encountering significant error when the camera is above the target object. This is an issue as the camera will be mounted on a drone. We think it's because the streams are enabled to 640x480 (4:3) resolution but the FOV values we're using (69.4, 42.5) are meant for 16:9 aspect ratio. When I try to change the resolution (this code is based off the python example in the SDK concerned with aligning depth to color), I get the following error: 

Traceback (most recent call last):                                                                                                                                                                                                             File "rs-detection.py", line 17, in <module>                                                                                                                                                                                                   profile = pipeline.start(config)                                                                                                                                                                                                         RuntimeError: Couldn't resolve requests    

This portion of the code causes the error:
`pipeline = rs.pipeline() 
config = rs.config() 

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

config.enable_stream(rs.stream.depth, 1920, 1080, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1920, 1080, rs.format.bgr8, 30)

profile = pipeline.start(config)

depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth scale: ", depth_scale)

align_to = rs.stream.color
align = rs.align(align_to) 

fov_x = 69.4
fov_y = 42.5
last_updated_dist = 0
last_updated_height = 0`

And this code works:
`pipeline = rs.pipeline() 
config = rs.config() 

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

config.enable_stream(rs.stream.depth, 1920, 1080, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1920, 1080, rs.format.bgr8, 30)

profile = pipeline.start(config)

depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth scale: ", depth_scale)

align_to = rs.stream.color
align = rs.align(align_to) 

fov_x = 69.4
fov_y = 42.5
last_updated_dist = 0
last_updated_height = 0`

Any help on this would be appreciated!

MartyG-RealSense, 2021-04-04:
Hi @pranav-pang   The error **RuntimeError: Couldn't resolve requests** is occurring because you are requesting a depth resolution that the camera does not support.  The maximum depth resolution available is **1280x720**.

[image]

pranav-pang, 2021-04-04:
Lowering the resolution doesn't seem to work either:

`w_max = 1280
h_max = 720

config.enable_stream(rs.stream.depth, w_max, h_max, rs.format.z16, 30)
config.enable_stream(rs.stream.color, w_max, h_max, rs.format.bgr8, 30)

profile = pipeline.start(config)`

I still receive the same error, so maybe I'm not quite understanding some issue here

MartyG-RealSense, 2021-04-04:
Is your camera on a USB 2 connection please?  If it is then you will only be able to access a limited range of resolutions and FPS speeds.  For example, on USB 2 1280x720 is supported but only 6 FPS.

You should be able to use 640x480 at 30 FPS on USB 2.  The link below containers chart of supported USB 2 resolutions and the available speeds for each resolution.  Calling a mode not supported on USB 2 will result also in **RuntimeError: Couldn't resolve requests**

closed 2021-04-17 by MartyG-RealSense