realsenseai/librealsense #8736: Rs-Enumerate does not give units with extrinsics

geoeo, 2021-04-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 2.41 | 
| Operating System & Version |   {Win 10| 
| Platform                                 | PC/ |
| SDK Version                            |  { 2.4}                          | 
| Language                            |  compiled tool                        | 
| Segment			|  photogrammetry                       | 

### Issue Description
The rs-enumerate tool gives the extrinsics between sensors, which is great(!), but does not give units (which is not great!).
Its really important to give the user this information as well otherwise the whole procedure is useless.

Is it meters, cm? pixels?

Best,
Marc

MartyG-RealSense, 2021-04-04:
Hi @geoeo   Rotation is in the form of a 3x3 rotation matrix, and the translation between physical sensors is expressed in the **meters** unit.

MartyG-RealSense, 2021-04-11:
Hi @geoeo  Do you require further assistance with this case, please?  Thanks!

geoeo, 2021-04-11:
It would also be nice to have an image showing the axes. Theres something floating around for the d435i, but im not sure its official.

MartyG-RealSense, 2021-04-11:
I recalled that I had provided a link to such a diagram before.  It is in the link below.  As you say though, it is not official.

https://github.com/IntelRealSense/librealsense/issues/7568#issuecomment-708358084

There is an official D435i illustrated description of the relationship between IMU and depth sensor in the Intel IMU guide linked to below.

https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/

[image]

geoeo, 2021-04-12:
yes something like 
[image]

for the realsense would be nice

MartyG-RealSense, 2021-04-12:
I have added a **Documentation** tag to this case so that your request for an official diagram of the translation between all sensors can be considered by the RealSense team.  This case should be kept open in the meantime.  Thanks!

geoeo, 2021-04-17:
Reopened this as per your comment. Thanks!

closed 2026-10-23 by MartyG-RealSense