realsenseai/librealsense #8739: OpenNI2 always reports same FoV regardless of actual resolution/aspect ration

noirflux, 2021-04-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435, D455 | 
| Operating System & Version |   Win7, Win10  | 
 Platform                                 | PC |
| SDK Version                            |  2.42, 2.43                         | 
| Language                            |  C# via NiWrapper.Net                          | 

### Issue Description
OpenNI2 is reporting the same FoV on the depth stream no matter which resolution is set, even 848x100:
Width: 1.39665293693542   Height: 1.12380337715149
Looks like a 4:3 aspect to me.

OpenNI is initialized, the devices enumerated, the device is opened, the depth video stream created, its video mode set, and then the depth stream FoV read and then resolution read.  The resolution is reported correctly in all cases, the FoV is always the same.

In a related area, the Intel docs for the D400 series just list one set of FoVs.  I assume this is for 16:9 settings; would be nice if they listed them for all the various resolutions.

MartyG-RealSense, 2021-04-05:
Hi @noirflux  You can check the current FOV values in C# using code from the SDK's **C# Cookbook** page of example code:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#get-field-of-view

noirflux, 2021-04-05:
Thanks.  But how is the OpenNI2 wrapper doing it?  Doing a little code archeology, the FoV seems to be retrieved in Rs2Stream.cpp.  startPipeline() calls onStreamStarted() which calls updateInstrinsics(), which does:

	Rs2StreamProfileInfo* spi = getCurrentProfile();
	rs2_get_video_stream_intrinsics(spi->profile, &m_intrinsics, nullptr);
	rs2_fov(&m_intrinsics, &m_fovX);

(BTW referencing consecutive distinct elements in a struct/class as is done with the &m_fovX instead of the data being in a struct or array is an invitation to future bugs when some oblivious programmer changes that order!  rs2_fov() treats that as an array reference with the assumption that m_fovX[1] is going to be the Y element.  That should be written at least as rs2_fov(&m_intrinsics, &m_fovX, &m_fovY).  But back to the chase...)

In that same file in Rs2Stream::initialize(), there are some scary TODO comments where defaults of 640x480 are set and then updateInstrinsics() is called.  That could explain the observed 4:3 FoVs; are the intrinsics always being updated and updateInstrinsics() always being called when it should, like after a resolution is set?  Thanks!

MartyG-RealSense, 2021-04-06:
I do not have any references about how the OpenNI2 wrapper handles FOV other that your own information above, so I do not have advice that I can offer on that subject unfortunately.

MartyG-RealSense, 2021-04-12:
Hi @noirflux   Do you have further questions about this case, please?  Thanks!

noirflux, 2021-04-12:
Well it is still an issue.   As the OpenNI2 wrapper is a part of this librealsense project here, who then should be looking into this?  This and some of the other issues I am seeing make it look like the OpenNI2 wrapper was only tested (and maybe only designed) for Primesense 1080 compatibility.  I do not have the familiarity with the wrapper or librealsense internals to delve into it to figure that out, so who does?

How is the wrapper unit tested/QA'd?  Unfortunately the examples with OpenNI2 only exercise a small part of the API.  Thanks.

MartyG-RealSense, 2021-04-12:
The OpenNI2 wrapper was contributed by an outside group in September 2018 and has had few updates since then.  So I believe that it should be considered to be in an "As Is" state that works and s maintained in that working state but may not receive many future updates (the most recent ocntribution was December 2019).

noirflux, 2021-04-12:
Ahhh, I see.  Unfortunately I am not familiar with the RealSense SDK API at all, so not sure how much help I can be.  But a couple of these things may be painful enough (like the Field of View issue) that just delving in and fixing it is probably easier than special-casing it in my code.  I of course will contribute any such fixes.  Like say an update of the OpenNI2 viewer that actually allows you to set res, FPS, etc. for testing!

MartyG-RealSense, 2021-04-12:
Thanks very much @noirflux  :)

closed 2021-04-26 by MartyG-RealSense