realsenseai/librealsense #8740: Question  SR300/SR305 filter option: Whicht kind/family/type of filter

hernot, 2021-04-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 19.04) | 
| Kernel Version (Linux Only)    |  (e.g. 3.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.x.x }                          | 
| Language                            |  python/unity                           | 
| Segment			|  others                           | 

### Issue Description

Just wondering what kind/type of filter is represented by each filter option value with 0 meaning no filter(off)?
It is sufficient for me to know which kind/type of filter is applied at which stage of processing ba the device.

Is it some sort of blurring filter applied to IR stream before extracting depth values and option selects kernel size (eg: 1->3x3, 2 ->5x5 etc) or is it that each option selects a different filter and/or stream applied to eg (1:gauss 3x3 ir stream , 2 gauss 5x5 IR stream , 3 edge-enhacement 3x3 ir stream, ..., 6 gauss 3x3 depth stream ...)?

A very general spec would already be sufficient. In case that would  not be possible without revealing secrets about the inner working of the device, alternatively a short description of the application each option is intended for and a reference to examples where its effect can be judged from would also be fine.

MartyG-RealSense, 2021-04-05:
Hi @hernot   As far as I know, none of the above applies to the SR300 / SR305 camera models.

The data sheet document for these models - based on **coded light** technology instead of the **stereo depth** of the 400 Series models - provides the following description for how the depth frame is generated.

***************

The Intel RealSense Depth Camera SR300 series depth camera uses multiple components to generate a depth image.  To generate a depth frame, the IR projector illuminates the scene with a set of predefined, increasing spatial frequency coded IR vertical bar patterns.  These patterns are warped by the scene, reflected back and captured by the IR camera.  The IR camera pixel values are then processed by the vision processor to generate a depth frame.  Subsequent depth frames create a video
stream that is transmitted to the host system.

***********

The data sheet is available from the link below:

https://dev.intelrealsense.com/docs/intel-realsense-sr300-datasheet

The technical information for the infrared sensor on page 15 of the data sheet states that the sensor has an **IR bandpass filter**.

[image]

In regard to the camera's distortion model, the **Projection** documentation states that the SR300 depth and infrared streams use the **Inverse Brown-Conrady** model.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#distortion-models

The same documentation page describes model-specific projection details for the SR300.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#appendix-model-specific-details

hernot, 2021-04-05:
So does the SR2_OPTION_FILTER_OPTION option change the characteristics of this ir Bandpass filter or does the vision processor apply an additional digital filter to the IR / depth stream which can be selected out of up to 7 possible presets or turned off by setting option to 0 that is what my question is about and is neither answered by the data sheet there it just says see filter option settings nor by your other links.

MartyG-RealSense, 2021-04-05:
The description for RS2_OPTION_FILTER_OPTION is "Set the filter to apply to each depth frame.  Each one of the filter is optimized per the application requirements".  The table below from the original June 2016 edition of the SR300 data sheet describes what the 0-7 settings correspond to.

[image]

hernot, 2021-04-05:
Thank you very much that is missing from current data sheet. Just to sum up and not misinterpret the table hi fidelity means only those pixel values for which depth value is highly certain. Gradient filter and sharpness mean some gradient based image enhancement filter and smoothing filters are sort of blurring filter to smooth out undesired pixel. Just that both kinds of filters are optimized for images representing depth instead of luminance or greyscale image. Even in case this naive interpretation should not be proper it at least fits my subjective empircal observations when switching between the different filters. Thanks again.

closed 2021-04-05 by hernot

MartyG-RealSense, 2021-04-06:
You are very welcome @hernot - thanks for the update!