realsenseai/librealsense #8741: D455 Tilt Angle with Python

Somusundram, 2021-04-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 }                                | 
| Firmware Version                   | {05.12.09.00}                         | 
| Operating System & Version | { Win (8.1/10)}                       | 
| Kernel Version (Linux Only)   | {}                                            |                                
| Platform                                 | {PC}                                       |
| SDK Version                           | {pyrealsense2 2.43.0.3018}   |  
| Language                               | {Python}                               | 
| Segment			         | {others}                                | 

Hi.

I have tried to get tilt angle from IMU under Python. However i am unable to do so. I have referred  **rs-motion** but not clear on how to read constant tilt angle (x ,y ,z) from continues frames.

Correct me if i am wrong. Thanks

MartyG-RealSense, 2021-04-05:
Hi @Somusundram   If you are seeking to obtain real-time values from the IMU for Roll, Pitch and Yaw, I hope that the discussion in the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/4391

Somusundram, 2021-04-05:
@MartyG-RealSense 

Thanks for the point out. However i am stuck at this part:

`def accel_data(accel):`
`    return np.asarray([accel.x, accel.y, accel.z])`

`                if frame.is_motion_frame():`
`                    accel = accel_data(frame.as_motion_frame().get_motion_data())`

How to get gyro data from here?

MartyG-RealSense, 2021-04-05:
Can you try this please to get gyro:

**gyro = gyro_data(frame.as_motion_frame().get_motion_data())**

MartyG-RealSense, 2021-04-12:
hi @Somusundram  Do you require further assistance with this case, please?  Thanks!

Somusundram, 2021-04-13:
@MartyG-RealSense  Thanks. Though this is not the one i am looking for, i manage to get the desired result which helped tremendously on understanding its feature. I will close this ticket. Thank you.

closed 2021-04-13 by Somusundram

MartyG-RealSense, 2021-04-13:
Great to hear, @Somusundram - thanks very much for the update!