realsenseai/librealsense #8742: error: no matching function for call to ‘rs2::software_sensor::software_sensor()’

sandeshk1, 2021-04-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version | Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |  5.4                                  | 
| Platform                                 | core i7 PC  |
| SDK Version                            |  2.41.0                           | 
| Language                            |  C/C++         | 
| Segment			|  Computer Vision                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am trying to define a sensor by following the rs-software-device.cpp code but getting the following error while declaring the sensor object.

    error: no matching function for call to ‘rs2::software_sensor::software_sensor()’
    rs2::software_sensor depth_sensor;

Note that in the example code "auto" keyword is used to specify the type but in my application I need to declare the variable with the correct type (https://github.com/IntelRealSense/librealsense/blob/5ff27fca5aaeec4736d6bb3bfb958fee82ee198b/examples/software-device/rs-software-device.cpp#L138)

MartyG-RealSense, 2021-04-05:
Hi @sandeshk1   A script in the link below takes the same approach with **software_sensor depth_sensor;** that you have.

https://stackoverflow.com/questions/52585921/convert-opencv-mat-to-frame-type-of-realsense-in-librealsense-sdk

Did you define **rs2::software_device dev** in the line before your software_sensor instruction, please?

sandeshk1, 2021-04-05:
Hi @MartyG-RealSense Yes, **rs2::software_device dev** was defined in the line before. I am also able to define **rs2::stream_profile depth_stream**. But unfortunately defining **rs2::software_sensor depth_sensor** throws a compilation error

MartyG-RealSense, 2021-04-06:
Would it be possible to post your script in this discussion please (or at least the opening section of it if it is very long)?  Thank you.

MartyG-RealSense, 2021-04-12:
Hi @sandeshk1  Do you require further assistance with this case, please?  

If you do, please see the comment above in regard to posting a script.  Thanks!

sandeshk1, 2021-04-15:
@MartyG-RealSense  sorry for the delayed response

I have just modified the rs-imshow.cpp code of librealsense example 

    #include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
    #include <librealsense2/hpp/rs_internal.hpp>
    #include <opencv2/opencv.hpp>   // Include OpenCV API
    
    int main(int argc, char * argv[]) try
    {
        // Declare depth colorizer for pretty visualization of depth data
        rs2::colorizer color_map;
    
        // Declare RealSense pipeline, encapsulating the actual device and sensors
        rs2::pipeline pipe;
        // Start streaming with default recommended configuration
        pipe.start();
    
        using namespace cv;
        const auto window_name = "Display Image";
        namedWindow(window_name, WINDOW_AUTOSIZE);
    
        const int W = 848;
        const int H = 480;
        const int BPP = 2;
    
        int frame_number = 0;
    
        struct synthetic_frame
        {
            int x, y, bpp;
            std::vector<uint8_t> frame;
        };
    
    
        synthetic_frame depth_frame;
        synthetic_frame color_frame;
    
        depth_frame.x = W;
        depth_frame.y = H;
        depth_frame.bpp = BPP;
    
        color_frame.x = W;
        color_frame.y = H;
        color_frame.bpp = BPP;
    
    
        rs2_intrinsics color_intrinsics = { color_frame.x, color_frame.y,
            (float)color_frame.x / 2, (float)color_frame.y / 2,
            (float)color_frame.x / 2, (float)color_frame.y / 2,
            RS2_DISTORTION_BROWN_CONRADY ,{ 0,0,0,0,0 } };
    
        rs2_intrinsics depth_intrinsics = { depth_frame.x, depth_frame.y,
            (float)depth_frame.x / 2, (float)depth_frame.y / 2,
            (float)depth_frame.x , (float)depth_frame.y ,
            RS2_DISTORTION_BROWN_CONRADY ,{ 0,0,0,0,0 } };
    
        rs2::software_device dev; // Create software-only device
    
        auto depth_sensor = dev.add_sensor("Depth"); // Define single sensor
        auto color_sensor = dev.add_sensor("Color"); // Define single sensor
    
        auto depth_stream = depth_sensor.add_video_stream({  RS2_STREAM_DEPTH, 0, 0,
                                    W, H, 60, BPP,
                                    RS2_FORMAT_Z16, depth_intrinsics });
    
        depth_sensor.add_read_only_option(RS2_OPTION_DEPTH_UNITS, 0.001f);
    
    
        auto color_stream = color_sensor.add_video_stream({  RS2_STREAM_COLOR, 0, 1, W,
                                    H, 60, BPP,
                                    RS2_FORMAT_RGB8, color_intrinsics });
    
        dev.create_matcher(RS2_MATCHER_DLR_C);
        rs2::syncer sync;
    
        depth_sensor.open(depth_stream);
        color_sensor.open(color_stream);
    
        depth_sensor.start(sync);
        color_sensor.start(sync);
    
        depth_stream.register_extrinsics_to(color_stream, { { 1,0,0,0,1,0,0,0,1 },{ 0,0,0 } });
    
        while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
        {
            rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
            rs2::frame depth = data.get_depth_frame();
            rs2::frame color = data.get_color_frame();
    
            // Query frame size (width and height)
            const int w = depth.as<rs2::video_frame>().get_width();
            const int h = depth.as<rs2::video_frame>().get_height();
            const int cw = color.as<rs2::video_frame>().get_width();
            const int ch = color.as<rs2::video_frame>().get_height();
    
    
            // Create OpenCV matrix of size (w,h) from the colorized depth data
            Mat image(Size(w, h), CV_16U, (void*)depth.get_data(), Mat::AUTO_STEP);
            Mat cimage(Size(cw, ch), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);
    
            depth_sensor.on_video_frame({ image.data, // Frame pixels from capture API
                [](void*) {}, // Custom deleter (if required)
                depth_frame.x*depth_frame.bpp, depth_frame.bpp, // Stride and Bytes-per-pixel
                (rs2_time_t)frame_number * 16, RS2_TIMESTAMP_DOMAIN_HARDWARE_CLOCK, frame_number, // Timestamp, Frame# for potential sync services
                depth_stream });
    
    
            /*color_sensor.on_video_frame({ cimage.data, // Frame pixels from capture API
                [](void*) {}, // Custom deleter (if required)
                color_frame.x*color_frame.bpp, color_frame.bpp, // Stride and Bytes-per-pixel
                (rs2_time_t)frame_number * 16, RS2_TIMESTAMP_DOMAIN_HARDWARE_CLOCK, frame_number, // Timestamp, Frame# for potential sync services
                color_stream });
            */
    
            ++frame_number;
    
            rs2::frameset fset = sync.wait_for_frames();
            rs2::depth_frame rs2_depth = fset.first_or_default(RS2_STREAM_DEPTH);
    
            Mat sim_image(Size(848, 480), CV_16U, (void*)rs2_depth.get_data(), Mat::AUTO_STEP);
    
            //rs2::frame rs2_color = fset.first_or_default(RS2_STREAM_COLOR);
    
            std::cout << "rs2_depth units: " << rs2_depth.get_units() << "\n";
            std::cout << "rs2_depth get_distance: " << rs2_depth.get_distance(200, 200) << "\n";
    
            // Update the window with new data
            imshow(window_name, image);
        }
    
        return EXIT_SUCCESS;
    }

I cannot declare depth_sensor as `rs2::software_sensor depth_sensor; ` to be able to use as a private member variable in my application.

In the example of rs_software_device.cpp auto keyword is used to define the sensor

sandeshk1, 2021-04-15:
Also in the above code when I try to simulate color frame with color_sensor.on_video_frame function, I get the below error

    RealSense error calling rs2_get_frame_data(frame_ref:nullptr):
        null pointer passed for argument "frame_ref"

MartyG-RealSense, 2021-04-15:
I attempted to debug your problem but I do not have sufficient knowledge of the **software_device** system to do so unfortunately.  I do apologise.

MartyG-RealSense, 2021-04-21:
Hi @sandeshk1   Do you wish to contiue this case, please?  Thanks!

sandeshk1, 2021-04-21:
@MartyG-RealSense It is nice if we can have this issue resolved. For now I am following the workaround of declaring & assigning using auto keyword as a global variable instead of just declaring it as a private member

closed 2021-04-21 by sandeshk1